winder/Universal-G-Code-Sender #1287: Soft limits creates the cutting volume with a box in the visualizer window (see pictures). However, the box is in the 3rd quadrant and it should be in the first quadrant

rlwoodjr, 2019-10-16:
Soft limits creates the cutting volume with a box in the visualizer window (see pictures). However, the box is in the 3rd quadrant and it should be in the first quadrant. This box is not shown if $20=0. It is visible with $20=1.

The Z axis part of the box is correct.  X and Y need to be positive.

Homing the CNC does not correct the issue.

[image]
[image]

breiler, 2019-10-17:
What is your $23 setting (Homing invert)?

As I understood from the GRBL manual when implementing this, it will try to home the machine by moving to the furthest axis positive direction. However I don't take the homing invert mask into account:
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#23---homing-dir-invert-mask

The $20 enable/disable setting is correct. I could use the settings for soft limit max without enabling soft limits. It just didn't feel intuitive, but it can easily be changed.

breiler, 2019-10-17:
Yes, it's possible and I think I've already solved it. But I need to verify the new code using my homing-test-rig which is in a box somewhere in the basement. I'll see if get some time for this tonight.

Reading my comment again I realized it made no sense... 😄 
> I could use the settings for soft limit max without enabling soft limits. It just didn't feel intuitive, but it can easily be changed.

So the question is, when should the machine boundary be visible in the visualizer? 
Right now it's only activated when soft limits are enabled ($20=1) like you mentioned. An other alternative is show it already when homing is enabled. 

We would still use the same settings for the size of bounding box that is used for defining the soft limits: (https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#130-131-132--xyz-max-travel-mm)

But technically we don't need soft limits to be activated to show the box.

Which alternative would you say is the most intuitive?

rlwoodjr, 2019-10-17:
I like the idea of a box it can show our customers the cutting area. It would show that the tool path is in or out of bounds.  I would not display if $20=0, but we would never recommend $20=0, except for diagnosing issues.

 I suppose you may want to see the box on a CNC that does not home. Can you set soft limits if you do not home? 

Maybe a checkbox so the user can turn it on or off. 

A cool feature would be to turn the plane  causing the Alarm2 red so that the customer could see why he was getting the Alarm2   If you can read gcode the information is already there, but visual would be cool.

breiler, 2019-10-17:
The box can not be shown if homing isn't enables as there is no way of determining the machine position. So I'll leave it as it is, it will only be shown if soft limits are enabled (the box can be deactivated/hidden from the right-click menu in the visualizer).

It would indeed be cool to visualize if any toolpath in the gcode would cause the machine to exceed it's limits. I'll create another issue for this.

rlwoodjr, 2019-10-17:
Thanks! I appreciate the quick response.

closed 2019-10-17 by breiler

breiler, 2019-11-20:
@rlwoodjr, did you have the time to test this?

I managed to get my machine up and running with homing switches and tested this. I have both X and Y switches inverted and after homing my machine coordinates becomes my negative soft limits.

So with the changes I made, I get the machine bounding box in the wrong quadrant.

Did you run homing with your machine?

rlwoodjr, 2019-11-20:
Yes, the box is in the correct quadrant when I home.

PR #1288 (2019-10-17): Feature/handle inverted homing
Fixes #1287

As far as I've tested this should fix the visualization of the machine boundry if the homing direction is inverted. (It still requires that soft limits is enabled)

[image]