winder/Universal-G-Code-Sender #1291: Raspberry Pi Installation Page is out of date

MonkeyManGuy, 2019-10-23:
When following the Raspberry Pi installation page, numerous areas need to be updated as they no longer function:
1. Found on the installation page= Choose "A6 GL Driver"
The GL driver is now listed under option A7, and the language used on the resulting page does not match up with the guide. This makes it unclear how to "Choose "G1 GL (Full KMS)," which is the next step in the directions.

2. The command line: "apt-get install openjdk-8-jdk" no longer functions. It results in a message of having attempted to download the (what seems like) the correct software, but failed to do so.

[image]
[image]

breiler, 2019-10-26:
Thanks, i've updated the documentation. But I noticed that there's  now a bug in rasbian that will make it unresponsive if using GL with full KMS and a touchscreen. There's a fix for it waiting to be built and shipped by the rasbian team.

closed 2019-10-26 by breiler

MonkeyManGuy, 2020-01-23:
> Thanks, i've updated the documentation. But I noticed that there's now a bug in rasbian that will make it unresponsive if using GL with full KMS and a touchscreen. There's a fix for it waiting to be built and shipped by the rasbian team.

Breiler, thanks for the insight. Do you know if they have completed this fix? Im actually wanting to use a touch screen although I haven't gotten that far. 

I still cant get UGS working on my Pi. The install instructions still deviate from how the Pi responds after just a few steps.