winder/Universal-G-Code-Sender #1296: Importing keymapping crashes stable version

falviani, 2019-10-27:
## Feature request

## Problem description 

### Expected Behavior

### Actual Behavior

### Steps to Reproduce the Problem

  1.
  1.
  1.

### Specifications
#### Version

#### Operating system

#### Platform

#### Other

falviani, 2019-10-27:
I created a set of keymappings in the latest platform nightly and exported just them. I then  opened the latest platform stable and imported them. The app immediately crashed with no error messages being displayed.

BallscrewBob, 2019-11-07:
Good to see this brought up again as that's the issue I had.
In between some of the crashes it did actually import my macros and such so I never came back to the issue.

breiler, 2019-12-02:
The import/export function is unfortunately broken in earlier versions of UGS. This should have been fixed in #1156.

I'm closing this as there isn't much we can do with the old import files. =(

closed 2019-12-02 by breiler