winder/Universal-G-Code-Sender #1303: Wizard on calibration step

beckmx, 2019-11-11:
## Problem description 
In the latest stable build and also in the nightly build I noticed that if you want to use the wizard, in the step that let you calculate and configure the microstepping, somehow it doesnt display that section, it still shows the previous one, the one that let's you decide if the axis is inverted or not. I tested this on windows 10 and OSx

### Expected Behavior
The window or "screen" where you can calculate the steps should be shown

### Actual Behavior
The screen is not shown but the wizard says it's there

### Steps to Reproduce the Problem
With gbrl 1.1f open the Wizard setup and navigate trough the steps

### Specifications
#### Version

  UGS Platform 2.0 - November 2, 2019

#### Operating system

  Windows 10
  Mac OSX

#### Platform

  GRBL 1.1f

breiler, 2019-11-11:
Hmm, I can't reproduce this. Does this happen every time?

beckmx, 2019-11-11:
yes, I am not at home, but as soon as I get there I can attach you some pics

beckmx, 2019-11-13:
I recorded this little section of the issue https://www.youtube.com/watch?v=V-5rVBk9WSg&feature=youtu.be

breiler, 2019-11-13:
Thanks!

Could I have your log file from:
~/Library/Application\ Support/ugsplatform/2.0-SNAPSHOT/dev/var/log/messages.log

beckmx, 2019-11-13:
[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/3842700/messages.log)

breiler, 2019-11-13:
Ah, I've seen other issues with the similar error:
`java.lang.NoSuchMethodError: com.willwinder.universalgcodesender.model.Position.getX()D
	at com.willwinder.universalgcodesender.model.Position.get(Position.java:97)`

There is probably an old library file that is bundled with Java. One user reported that he/she removed to old library #261:
> To fix the getX() error I had to remove vecmath.jar from System/Library/Java/Extensions. This was an old version 1.3.

That seems a bit drastic though, @winder do you have any ideas?

breiler, 2019-11-25:
@beckmx if you have the time, you could give the edition with a bundled Java a try:
https://ugs.jfrog.io/ugs/UGS/nightly/ugs-platform-app-2.0-SNAPSHOT.dmg
https://ugs.jfrog.io/ugs/UGS/nightly/ugs-platform-app-2.0-SNAPSHOT-win.zip

winder, 2019-11-25:
@breiler vecmath has always been an issue, I think that accessing the properties directly (meaning `.X` instead of `.getX()`) is a portable fix. #1231 removes vecmath completely, but I don't know how stable the rest of those changes are so never merged it in.

breiler, 2020-01-07:
Closing as it's a configuration issue on MacOSX. This should work using UGS with bundled java.

closed 2020-01-07 by breiler