winder/Universal-G-Code-Sender #1304: Some potential NPE bugs

JulyChen728, 2019-11-12:
Hi all,

Our tool has found several potential NPE bugs

The first one is in [runSimulation](https://github.com/winder/Universal-G-Code-Sender/blob/v1.0.9/src/com/willwinder/universalgcodesender/GrblSimulator.java#L44).

Variable ```parserThread``` is initialized to null and will be assigned by ```parserThread = new Thread(simOutputBuffer);```. However, if the three instructions before the assignment raise an IOException, then the program will skip the assignment and execute  [```if (parserThread.isAlive())```](https://github.com/winder/Universal-G-Code-Sender/blob/v1.0.9/src/com/willwinder/universalgcodesender/GrblSimulator.java#L74).
Since variable ```parserThread``` is still null, then the program will raise a NullPointerException.

The second one is caused by the ```return null``` in method [center()](https://github.com/winder/Universal-G-Code-Sender/blob/v1.0.9/src/com/willwinder/universalgcodesender/types/PointSegment.java#L187).

For examples,
In [getLinesFromParser](https://github.com/winder/Universal-G-Code-Sender/blob/v1.0.9/src/com/willwinder/universalgcodesender/visualizer/GcodeViewParse.java#L129), it is a NPE bug if ```ps.center()``` return [null](https://github.com/winder/Universal-G-Code-Sender/blob/v1.0.9/src/com/willwinder/universalgcodesender/types/PointSegment.java#L187) and used in [center.x](https://github.com/winder/Universal-G-Code-Sender/blob/v1.0.9/src/com/willwinder/universalgcodesender/gcode/GcodePreprocessorUtils.java#L481)

If [```lastSegment.center()```](https://github.com/winder/Universal-G-Code-Sender/blob/v1.0.9/src/com/willwinder/universalgcodesender/gcode/GcodeParser.java#L167) returns null and used at [ GcodePreprocessorUtils.generatePointsAlongArcBDring](https://github.com/winder/Universal-G-Code-Sender/blob/v1.0.9/src/com/willwinder/universalgcodesender/gcode/GcodeParser.java#L175), [GcodePreprocessorUtils.getAngle](https://github.com/winder/Universal-G-Code-Sender/blob/v1.0.9/src/com/willwinder/universalgcodesender/gcode/GcodePreprocessorUtils.java#L484) and finally dereferenced at [end.y - start.y;](https://github.com/winder/Universal-G-Code-Sender/blob/src/com/willwinder/universalgcodesender/gcode/GcodePreprocessorUtils.java#L418)

The third one is caused by the ```return null``` in method [getControllerFor()](https://github.com/winder/Universal-G-Code-Sender/blob/v1.0.9/src/com/willwinder/universalgcodesender/utils/FirmwareUtils.java#L66).

For example, one call chain is [```this.controller = FirmwareUtils.getControllerFor(firmware);```](https://github.com/winder/Universal-G-Code-Sender/blob/src/com/willwinder/universalgcodesender/model/GUIBackend.java#L122), [```applySettingsToController(settings, this.controller);```](https://github.com/winder/Universal-G-Code-Sender/blob/src/com/willwinder/universalgcodesender/model/GUIBackend.java#L123) and [``` controller.getCommandCreator()...```](https://github.com/winder/Universal-G-Code-Sender/blob/src/com/willwinder/universalgcodesender/model/GUIBackend.java#L545)

Thanks.

breiler, 2019-11-12:
Nice, thanks.

Is there a way to run the tool on the master branch? The version 1.0.9 is quite old and some of the code that you are referencing has been removed.

JulyChen728, 2019-11-15:
> Nice, thanks.
> 
> Is there a way to run the tool on the master branch? The version 1.0.9 is quite old and some of the code that you are referencing has been removed.

Sure, here is the result on the master branch.

Null checks are inconsistent at lines [295](https://github.com/winder/Universal-G-Code-Sender/tree/54a1db413980d4c90b5b704c2723f0e740de20be/ugs-core/src/com/willwinder/universalgcodesender/BufferedCommunicator.java#L295) and [306](https://github.com/winder/Universal-G-Code-Sender/tree/54a1db413980d4c90b5b704c2723f0e740de20be/ugs-core/src/com/willwinder/universalgcodesender/BufferedCommunicator.java#L306), [100](https://github.com/winder/Universal-G-Code-Sender/tree/54a1db413980d4c90b5b704c2723f0e740de20be/ugs-core/src/com/willwinder/universalgcodesender/i18n/Localization.java#L100) and [101](https://github.com/winder/Universal-G-Code-Sender/tree/54a1db413980d4c90b5b704c2723f0e740de20be/ugs-core/src/com/willwinder/universalgcodesender/i18n/Localization.java#L101) (```region``` is used in the method ```initialize```).

In method ``` updateGLColorArray```, the null relations are also inconsistent. There is only one null check for ```lineColorData```  at line [387](https://github.com/winder/Universal-G-Code-Sender/tree/54a1db413980d4c90b5b704c2723f0e740de20be/ugs-platform/ugs-platform-visualizer/src/main/java/com/willwinder/ugs/nbm/visualizer/renderables/GcodeModel.java#L387), but the variable is used at lines [381](https://github.com/winder/Universal-G-Code-Sender/tree/54a1db413980d4c90b5b704c2723f0e740de20be/ugs-platform/ugs-platform-visualizer/src/main/java/com/willwinder/ugs/nbm/visualizer/renderables/GcodeModel.java#L381), [390](https://github.com/winder/Universal-G-Code-Sender/tree/54a1db413980d4c90b5b704c2723f0e740de20be/ugs-platform/ugs-platform-visualizer/src/main/java/com/willwinder/ugs/nbm/visualizer/renderables/GcodeModel.java#L390), [393](https://github.com/winder/Universal-G-Code-Sender/tree/54a1db413980d4c90b5b704c2723f0e740de20be/ugs-platform/ugs-platform-visualizer/src/main/java/com/willwinder/ugs/nbm/visualizer/renderables/GcodeModel.java#L393).

About methods that may return null, 
```getStateFromStatusString``` returns null at [460](https://github.com/winder/Universal-G-Code-Sender/tree/54a1db413980d4c90b5b704c2723f0e740de20be/ugs-core/src/com/willwinder/universalgcodesender/GrblUtils.java#L460), then the return value at line [340](https://github.com/winder/Universal-G-Code-Sender/tree/54a1db413980d4c90b5b704c2723f0e740de20be/ugs-core/src/com/willwinder/universalgcodesender/GrblUtils.java#L340) is used in [341](https://github.com/winder/Universal-G-Code-Sender/tree/54a1db413980d4c90b5b704c2723f0e740de20be/ugs-core/src/com/willwinder/universalgcodesender/GrblUtils.java#L341), finally, the null value may be dereferenced at line [473](https://github.com/winder/Universal-G-Code-Sender/tree/54a1db413980d4c90b5b704c2723f0e740de20be/ugs-core/src/com/willwinder/universalgcodesender/GrblUtils.java#L473)

Also, ```addLinearPointSegment``` may return null at line [289](https://github.com/winder/Universal-G-Code-Sender/tree/54a1db413980d4c90b5b704c2723f0e740de20be/ugs-core/src/com/willwinder/universalgcodesender/gcode/GcodeParser.java#L289), the method is called at [279](https://github.com/winder/Universal-G-Code-Sender/tree/54a1db413980d4c90b5b704c2723f0e740de20be/ugs-core/src/com/willwinder/universalgcodesender/gcode/GcodeParser.java#L279) and the null value may be used at [280](https://github.com/winder/Universal-G-Code-Sender/tree/54a1db413980d4c90b5b704c2723f0e740de20be/ugs-core/src/com/willwinder/universalgcodesender/gcode/GcodeParser.java#L280).

```lookupCode``` may return null at line [224](https://github.com/winder/Universal-G-Code-Sender/tree/54a1db413980d4c90b5b704c2723f0e740de20be/ugs-core/src/com/willwinder/universalgcodesender/gcode/util/Code.java#L224), then the method is called at [637](https://github.com/winder/Universal-G-Code-Sender/tree/54a1db413980d4c90b5b704c2723f0e740de20be/ugs-core/src/com/willwinder/universalgcodesender/gcode/GcodePreprocessorUtils.java#L637) and the null may be used at [638](https://github.com/winder/Universal-G-Code-Sender/tree/54a1db413980d4c90b5b704c2723f0e740de20be/ugs-core/src/com/willwinder/universalgcodesender/gcode/GcodePreprocessorUtils.java#L638).

```extractMotion``` may return null at lines [629](https://github.com/winder/Universal-G-Code-Sender/tree/54a1db413980d4c90b5b704c2723f0e740de20be/ugs-core/src/com/willwinder/universalgcodesender/gcode/GcodePreprocessorUtils.java#L629) and [646](https://github.com/winder/Universal-G-Code-Sender/tree/54a1db413980d4c90b5b704c2723f0e740de20be/ugs-core/src/com/willwinder/universalgcodesender/gcode/GcodePreprocessorUtils.java#L646), then the method can be called at line [81](https://github.com/winder/Universal-G-Code-Sender/tree/54a1db413980d4c90b5b704c2723f0e740de20be/ugs-core/src/com/willwinder/universalgcodesender/gcode/processors/LineSplitter.java#L81) and the null value may be used at line [82](https://github.com/winder/Universal-G-Code-Sender/tree/54a1db413980d4c90b5b704c2723f0e740de20be/ugs-core/src/com/willwinder/universalgcodesender/gcode/processors/LineSplitter.java#L82).

```fromString``` may return null at line [92](https://github.com/winder/Universal-G-Code-Sender/tree/54a1db413980d4c90b5b704c2723f0e740de20be/ugs-core/src/com/willwinder/universalgcodesender/utils/ControllerSettings.java#L92), then it is called at line [115](https://github.com/winder/Universal-G-Code-Sender/t [... cut ...] els/SendStatusPanel.java#L74), it calls ``` this(null);``` and variable ``` backend``` is set to null at line [78](https://github.com/winder/Universal-G-Code-Sender/tree/54a1db413980d4c90b5b704c2723f0e740de20be/ugs-core/src/com/willwinder/universalgcodesender/uielements/panels/SendStatusPanel.java#L78). Note that the variable ``` backend``` is frequently used in the class. Even though ``` SendStatusPanel()``` seems never been called, it might not be a good idea to keep this constructor.

Thanks.

breiler, 2019-11-15:
Excellent, my weekend is saved! 😁 
Thanks a bunch!

JulyChen728, 2019-11-18:
We also found a kind of potential ArrayIndexOutOfBounds bug.

The variable [lookups](https://github.com/winder/Universal-G-Code-Sender/blob/54a1db413980d4c90b5b704c2723f0e740de20be/ugs-core/src/com/willwinder/universalgcodesender/utils/GrblLookups.java#L58) is updated by ```lookups.put(record.get(0), list.toArray(new String[0]));```. However, the content in ```list``` is from a ```CSVRecord ``` without a file integrity check. If someone use a csv file with only two columns, then the array access at line [142](https://github.com/winder/Universal-G-Code-Sender/blob/54a1db413980d4c90b5b704c2723f0e740de20be/ugs-core/src/com/willwinder/universalgcodesender/GrblController.java#L142), [130](https://github.com/winder/Universal-G-Code-Sender/blob/54a1db413980d4c90b5b704c2723f0e740de20be/ugs-core/src/com/willwinder/universalgcodesender/types/GrblSettingMessage.java#L130) or [148](https://github.com/winder/Universal-G-Code-Sender/blob/54a1db413980d4c90b5b704c2723f0e740de20be/ugs-core/src/com/willwinder/universalgcodesender/firmware/grbl/GrblFirmwareSettingsCommunicatorListener.java#L148) will cause ArrayIndexOutOfBounds exception. 

Thanks.

ITWOI, 2019-11-21:
Hi, we have reported the bugs a while ago would you please take a look and confirm if they are real bugs. We have been conducting an experiment to measure the accuracy of our static checker. We would be deeply appreciated if you can provide some feedback!

breiler, 2019-11-21:
I haven't had the time to look in detail. Some of these are quite obvious a risk and should probably throw an exception or return optionals instead of null. I might get some time this weekend.

breiler, 2020-01-08:
Sorry for taking so long getting back to you guys! I've been looking through the code and verified problems on most of the places you pointed out.

What tool are you working on?

---
> Null checks are inconsistent at lines [...]

The null check is inconsistent but should never be a problem as the list is always created in the default constructor. We do however allow a null list to be assigned which should probably be asserted and throw an illegal argument exception.

---

> Null checks are inconsistent at lines [100](https://github.com/winder/Universal-G-Code-Sender/tree/54a1db413980d4c90b5b704c2723f0e740de20be/ugs-core/src/com/willwinder/universalgcodesender/i18n/Localization.java#L100) and [101](https://github.com/winder/Universal-G-Code-Sender/tree/54a1db413980d4c90b5b704c2723f0e740de20be/ugs-core/src/com/willwinder/universalgcodesender/i18n/Localization.java#L101) (```region``` is used in the method ```initialize```).

Yep, this one was not good. =)

---

> In method ``` updateGLColorArray```, the null relations are also inconsistent. There is only one null check for ```lineColorData```  at line [...]

I had a hard time following all the code in this. We should probably look further into this

---

> ```getStateFromStatusString``` returns null at [...]

This should be handled differently even though I think it's not likely it would happen. The solution here is to return a "unknown" status string instead of null.

---

> ```addLinearPointSegment``` may return null at line [...]

This might very well lead to NPE:s and should be fixed!

---

> ```lookupCode``` may return null at line [...]

Yep, this should probably not return null. It didn't look like the code using this methods would pass any unwanted codes to it. But to reduce the risk of introducing bugs on new code it should probably handle this differently.

---

> ```extractMotion``` may return null at lines [...]

Yes this seems like a risk.

---

> ```fromString``` may return null at line [...]

This is very unlikely it will ever happen, but should of course be fixed.

---

> ```getSettingsTextField``` may return null at [...]

This would very likely cause problems when we add additional axises. Thanks!

---

> ```lookup``` may return null at [...]

This looks like a low risk and is likely never to happen, but should of course be fixed.

---

> An interesting one is at method ``` SendStatusPanel()``` at line [...]

Yes, that shouldn't work... =)