winder/Universal-G-Code-Sender #1309: RaspPi build from Snapshot missing nbm-maven-plugin:4.2:manifest

KineticSculpt, 2019-11-16:
## Feature request

## Problem description 
 Walked through Rasp Pi setup instructions many times, seemed to have new failures every time except for repeating latest. I'm stuck at this.. missing plug in? 

### Expected Behavior
Set up a running UGS platform

### Actual Behavior
Missing.. something?

### Steps to Reproduce the Problem

  1.  Followed instructions on: https://github.com/winder/Universal-G-Code-Sender/wiki/Raspberry-Pi
  1. Ran: mvn install -P '!create-pendant-web-ui'
  1.

### Specifications
#### Version
https://github.com/winder/Universal-G-Code-Sender/archive/master.zip

#### Operating system
Raspbian
Rasp Pi 3

#### Platform

#### Other

[ERROR] Project depends on packages not accessible at runtime in module com.willwinder:ugs-platform-ugslib:jar:2.0-SNAPSHOT
[INFO] ------------------------------------------------------------------------
[INFO] Reactor Summary for ugs-parent 2.0-SNAPSHOT:
[INFO]
[INFO] ugs-parent ......................................... SUCCESS [  3.335 s]
[INFO] ugs-core ........................................... SUCCESS [02:15 min]
[INFO] ugs-pendant ........................................ SUCCESS [ 12.362 s]
[INFO] ugs-cli ............................................ SUCCESS [  1.621 s]
[INFO] ugs-classic ........................................ SUCCESS [ 21.473 s]
[INFO] ugs-platform-parent ................................ SUCCESS [  1.120 s]
[INFO] ugs-platform-branding .............................. SUCCESS [ 16.201 s]
[INFO] ugs-platform-ugslib ................................ SUCCESS [01:05 min]
[INFO] ugs-platform-ugscore ............................... SUCCESS [ 22.596 s]
[INFO] ugs-platform-visualizer ............................ SUCCESS [ 23.262 s]
[INFO] ugs-platform-plugin-workflow ....................... SUCCESS [  8.671 s]
[INFO] ugs-platform-gcode-editor .......................... SUCCESS [ 23.962 s]
[INFO] ugs-platform-surfacescanner ........................ SUCCESS [  7.474 s]
[INFO] ProbeModule ........................................ SUCCESS [ 18.352 s]
[INFO] DowelModule ........................................ SUCCESS [01:07 min]
[INFO] GcodeTools ......................................... SUCCESS [  7.671 s]
[INFO] ugs-platform-plugin-jog ............................ FAILURE [ 13.280 s]
[INFO] ugs-platform-plugin-setup-wizard ................... SKIPPED
[INFO] ugs-platform-welcome-page .......................... SKIPPED
[INFO] ugs-platform-app ................................... SKIPPED
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time:  07:42 min
[INFO] Finished at: 2019-11-15T21:28:08-08:00
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal org.apache.netbeans.utilities:nbm-maven-plugin:4.2:manifest (default-manifest) on project ugs-platform-plugin-jog: See above for failures in runtime NetBeans dependencies verification. -> [Help 1]
org.apache.maven.lifecycle.LifecycleExecutionException: Failed to execute goal org.apache.netbeans.utilities:nbm-maven-plugin:4.2:manifest (default-manifest) on project ugs-platform-plugin-jog: See above for failures in runtime NetBeans dependencies verification.
    at org.apache.maven.lifecycle.internal.MojoExecutor.execute (MojoExecutor.java:215)
    at org.apache.maven.lifecycle.internal.MojoExecutor.execute (MojoExecutor.java:156)
    at org.apache.maven.lifecycle.internal.MojoExecutor.execute (MojoExecutor.java:148)
    at org.apache.maven.lifecycle.internal.LifecycleModuleBuilder.buildProject (LifecycleModuleBuilder.java:117)
    at org.apache.maven.lifecycle.internal.LifecycleModuleBuilder.buildProject (LifecycleModuleBuilder.java:81)
    at org.apache.maven.lifecycle.internal.builder.singlethreaded.SingleThreadedBuilder.build (SingleThreadedBuilder.java:56)
    at org.apache.maven.lifecycle.internal.LifecycleStarter.execute (LifecycleStarter.java:128)
    at org.apache.maven.DefaultMaven.doExecute (DefaultMaven.java:305)
    at org.apache.maven.DefaultMaven.doExecute (DefaultMaven.java:192)
    at org.apache.maven.DefaultMaven.execute (DefaultMaven.java:105)
    at org.apache.maven.cli.MavenCli.execute (MavenCli.java:956)
    at org.apache.maven.cli.MavenCli.doMain (MavenCli.java:288)
    at org.apache.maven.cli.MavenCli.main (MavenCli.java:192)
    at jdk.internal.reflect.NativeMethodAccessorImpl.invoke0 (Native Method)
    at jdk.internal.reflect.NativeMethodAccessorImpl.invoke (NativeMethodAccessorImpl.java:62)
    at jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke (DelegatingMethodAccessorImpl.java:43)
    at java.lang.reflect.Method.invoke (Method.java:566)
    at org.codehaus.plexus.classworlds.launcher.Launcher.launchEnhanced (Launcher.java:282)
    at org.codehaus.plexus.classworlds.launcher.Launcher.launch (Launcher.java:225)
    at org.codehaus.plexus.classworlds.launcher.Launcher.mainWithExitCode (Launcher.java:406)
    at org.codehaus.plexus.classworlds.launcher.Launcher.main (Launcher.java:347)
Caused by: org.apache.maven.plugin.MojoFailureException: See above for failures in runtime NetBeans dependencies verification.
    at org.apache.netbeans.nbm.NetBeansManifestUpdateMojo.checkModuleClassPath (NetBeansManifestUpdateMojo.java:759)
    at org.apache.netbeans.nbm.NetBeansManifestUpdateMojo.execute (NetBeansManifestUpdateMojo.java:550)
    at org.apache.maven.plugin.DefaultBuildPluginManager.executeMojo (DefaultBuildPluginManager.java:137)
    at org.apache.maven.lifecycle.internal.MojoExecutor.execute (MojoExecutor.java:210)
    at org.apache.maven.lifecycle.internal.MojoExecutor.execute (MojoExecutor.java:156)
    at org.apache.maven.lifecycle.internal.MojoExecutor.execute (MojoExecutor.java:148)
    at org.apache.maven.lifecycle.internal.LifecycleModuleBuilder.buildProject (LifecycleModuleBuilder.java:117)
    at org.apache.maven.lifecycle.internal.LifecycleModuleBuilder.buildProject (LifecycleModuleBuilder.java:81)
    at org.apache.maven.lifecycle.internal.builder.singlethreaded.SingleThreadedBuilder.build (SingleThreadedBuilder.java:56)
    at org.apache.maven.lifecycle.internal.LifecycleStarter.execute (LifecycleStarter.java:128)
    at org.apache.maven.DefaultMaven.doExecute (DefaultMaven.java:305)
    at org.apache.maven.DefaultMaven.doExecute (DefaultMaven.java:192)
    at org.apache.maven.DefaultMaven.execute (DefaultMaven.java:105)
    at org.apache.maven.cli.MavenCli.execute (MavenCli.java:956)
    at org.apache.maven.cli.MavenCli.doMain (MavenCli.java:288)
    at org.apache.maven.cli.MavenCli.main (MavenCli.java:192)
    at jdk.internal.reflect.NativeMethodAccessorImpl.invoke0 (Native Method)
    at jdk.internal.reflect.NativeMethodAccessorImpl.invoke (NativeMethodAccessorImpl.java:62)
    at jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke (DelegatingMethodAccessorImpl.java:43)
    at java.lang.reflect.Method.invoke (Method.java:566)
    at org.codehaus.plexus.classworlds.launcher.Launcher.launchEnhanced (Launcher.java:282)
    at org.codehaus.plexus.classworlds.launcher.Launcher.launch (Launcher.java:225)
    at org.codehaus.plexus.classworlds.launcher.Launcher.mainWithExitCode (Launcher.java:406)
    at org.codehaus.plexus.classworlds.launcher.Launcher.main (Launcher.java:347)
[ERROR]
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR]
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoFailureException
[ERROR]
[ERROR] After correcting the problems, you can resume the build with the command
[ERROR]   mvn <goals> -rf :ugs-platform-plugin-jog

-----
Related note: Is there a Rasp Pi img with a relatively working UGS I missed somewhere? That would be really neat.

breiler, 2019-11-16:
I would just download the package, unzip and start:
https://github.com/winder/Universal-G-Code-Sender/wiki/Raspberry-Pi#download-and-start-ugs

I've also created a new package which bundles Java which could make things a bit simpler:
Simply download and extract: https://ugs.jfrog.io/ugs/UGS/nightly/ugs-platform-app-2.0-SNAPSHOT-pi.tar.gz

KineticSculpt, 2019-11-16:
Thank you for taking the time, breiler! I was able to successfully load UGS on the Pi. Many thanks!

closed 2019-11-16 by KineticSculpt