winder/Universal-G-Code-Sender #1312: Center Probe Code

dfox1787, 2019-11-18:
Hi guys

Does anyone have any code to find the center of a Hole? Im using a probe that i want to use to find a hole center?

[image]

breiler, 2019-11-19:
You could do this manually by using the expressions feature in the DRO:
http://winder.github.io/ugs_website/guide/platform/#digital-read-out

1. Go to the left of the hole, reset X
2. Go to the right of the hole, type "/2" in the field for the X work coordinate in the DRO 
3. Go to the X zero
4. Go to the bottom of the hole, reset Y
5. Go to the top of the hole, type "/2" in the field for the Y work coordinate in the DRO
6. Go to the Y zero

This should be the center, but you could run the steps one more time for it to be exact.
There's a video link in #977 that describes this process.

dfox1787, 2019-11-19:
thanks. im looking to add this in as a macro if possible.

owenduffy, 2019-11-25:
I think that enhancement of the probe module to exploit the type of calibration fixture shown would be useful. It the plate is made to sufficient accuracy, the calibration is independent of the cutter size (using the small hole which is centred over the work zero).

risototh, 2023-04-09:
I'm currently working on this. I have 80% of code completed, just strugling with the OpenGL as I want to visualize the process - jsut show a block with the hole and arrows... Arrows are not a problem, but the solid block with hole seems to be a problem for me, for now, today :) But I'm prettu sure, that I will break it. Here is a preview https://youtu.be/gPf3ezeDDHY

owenduffy, 2023-04-09:
Hi,

I looked at your video, and it looks promising.

Might I suggest you consider the case where the hole is not much larger than the cutter diameter (eg a 5mm hole and 3mm cutter).

Accuracy might be improved if you:

1. probe X- to X+ to find the X centre;
2. move X to centre;
3. probe Y1 to Y+ to find Y centre;
4. move Y to centre;
5. now redo X (now the cutter is approximately centred);
6. now redo Y (now the cutter is approximately centred);

Are you considering including a Z touch off in the same process? You might set the cutter / probe above the Z surface and over the hole, lower the cutter by a specified amount to probe the hole, return to initial Z, move a given X,Y to perform the Z touch off.

Thanks
Owen

risototh, 2023-04-09:
Hi Owen,

in fact, this double strategy came to my mind :) But first, I want to get out this at least this "first try", and then maybe the double one as an option. Meanwhile, you can just run the probing twice - it would do the same, as the first run will position the probe to the calculated center...
Now I'm quite happy, as I finally managed the OpenGL - aaah, hate it :D https://youtu.be/ZbO8Fl-Avj8

risototh, 2023-04-09:
This is already working implementation https://youtu.be/dVpXjkA24Wg
Tomorrow I will review and cleanup the code and send the PR.

breiler, 2023-08-18:
Closing as the feature is now implemented in UGS 2.0.18 and higher

closed 2023-08-18 by breiler