winder/Universal-G-Code-Sender #1313: [Error] An error was detected while sending '0S30': error: Expected command letter. Streaming has been paused. **** The communicator has been paused ****

raimis170, 2019-11-24:
i need help to fix this error in UGS I create g code in inkscape and i did try with other codes too but nothing
in different versions of ugs i get same error.

[Error] An error was detected while sending '0S30': error: Expected command letter. Streaming has been paused.
**** The communicator has been paused ****

breiler, 2019-11-25:
I suspect that this is similar to: #603 

I'm guessing you are using a plugin in Inkscape that requires a modified version of GRBL that can handle a M300-command for lifting a pen with a servo.

To be able to help you we would need what software you are running:
* GRBL-version
* Arduino type (Due, Mega, other)
* Inkscape plugin
* UGS version

closed 2019-11-25 by raimis170

raimis170, 2019-11-25:
Fixed my problem

AVTG-1, 2023-08-06:
> I suspect that this is similar to: #603
>
> I'm guessing you are using a plugin in Inkscape that requires a modified version of GRBL that can handle a M300-command for lifting a pen with a servo.
>
> To [...]

DonVoid18, 2023-08-28:
@raimis170 disculpa, puedes compartir la solución

moshe582, 2025-01-15:
[falla en  imprimir  con universal.zip](https://github.com/user-attachments/files/18428886/falla.en.imprimir.con.universal.zip)

breiler, 2025-01-15:
@moshe582 read this: https://github.com/winder/Universal-G-Code-Sender/issues/603#issuecomment-383352765