winder/Universal-G-Code-Sender #1316: UGS 2.00 not starting

ee02004, 2019-12-03:
A few weeks ago I've downloaded and tested UGS 2.00. At the time it worked correctly. 
Last weekend I've tried to use the software and I wasn't able to use it because it doesn't start.
When I try to run the software on the command line I get the following messages:

C:\Users\RCGP>UniversalGcodeSender.json
The system cannot execute the specified program.

C:\Users\RCGP>start-windows.bat

C:\Users\RCGP>java -jar -Xmx256m UniversalGcodeSender.jar
dez 03, 2019 9:32:59 PM com.willwinder.universalgcodesender.utils.SettingsFactory loadSettings
INFO: Log location: C:\Users\RCGP\.ugs\UniversalGcodeSender.json
dez 03, 2019 9:32:59 PM com.willwinder.universalgcodesender.utils.SettingsFactory loadSettings
INFO: Loading settings.
Exception in thread "main" com.google.gson.JsonSyntaxException: java.lang.IllegalStateException: Expected BEGIN_OBJECT but was STRING at line 1 column 1
        at com.google.gson.internal.bind.ReflectiveTypeAdapterFactory$Adapter.read(ReflectiveTypeAdapterFactory.java:176)
        at com.google.gson.Gson.fromJson(Gson.java:803)
        at com.google.gson.Gson.fromJson(Gson.java:741)
        at com.willwinder.universalgcodesender.utils.SettingsFactory.loadSettings(SettingsFactory.java:59)
        at com.willwinder.universalgcodesender.MainWindow.<init>(MainWindow.java:111)
        at com.willwinder.universalgcodesender.MainWindow.main(MainWindow.java:218)
Caused by: java.lang.IllegalStateException: Expected BEGIN_OBJECT but was STRING at line 1 column 1
        at com.google.gson.stream.JsonReader.beginObject(JsonReader.java:374)
        at com.google.gson.internal.bind.ReflectiveTypeAdapterFactory$Adapter.read(ReflectiveTypeAdapterFactory.java:165)
        ... 5 more

I've already tried to run the UGS platform version and had a similar result. When I try to use it I can see that status bar moving, but when it reachs the 100% position the software crashes.
I'm using windows 10 home 64bits. On the same computer I can rung the UGS 1.09 version without problems.
Can you help?

breiler, 2019-12-04:
Sounds like your settings files are corrupt. Try following these instructions for clearing the settings:
https://github.com/winder/Universal-G-Code-Sender/wiki/Troubleshooting-Overview#recent-upgrade---reset-preferences

ee02004, 2019-12-04:
Thanks! Deleting the folder solved the problem!

closed 2019-12-04 by ee02004