winder/Universal-G-Code-Sender #1318: G2Core - comment remover breaks the code

hidoba, 2019-12-10:
## Feature request
Don't remove comments after M100, M100.1 and M101 commands for G2Core

## Problem description 
G2Core [has the syntax](https://github.com/synthetos/g2/wiki/Mcodes) that allows to change the internal parameters by M codes (M100, M100.1 and M101). For example, the following line:
`
M100 ({out1:1})
`
will switch the digital output _out1_ to 1.

However, UGS removes everything between the brackets, and the "Comment remover" option in the settings is greyed out, so the line loses meaning.

### Specifications
#### Version
UGS Platform 2.0 - Nightly Build Nov 15, 2019

#### Operating system
  Mac OSX

#### Platform
g2Core

hidoba, 2019-12-10:
Any temporary workaround?

breiler, 2019-12-10:
Wow, that was a new one I've never seen. ;)

You could try to deactivate the "Comment remover", similar to these instructions:
http://winder.github.io/ugs_website/guide/troubleshooting/#gcode-program-stopped-working

I'm not sure if I made it optional though for the g2core config. And I also don't know the side effects of disabling it as it will send comments to the controller. 

I might be able to give this a try as soon as I get home.

breiler, 2019-12-10:
Ah, bad luck. The feature is configured as non optional:
https://github.com/winder/Universal-G-Code-Sender/blob/master/ugs-core/src/resources/firmware_config/g2core.json

I can change this and make it optional so that you could give it a try.

hidoba, 2019-12-10:
thanks! it would be great if you could compile the modified version.
I'll disable it and make my own regular expression to remove comments. Or even try to send with comments.

breiler, 2019-12-10:
@winder this got a bit tricky... 

In [GCodeParserUtils](https://github.com/winder/Universal-G-Code-Sender/blob/master/ugs-core/src/com/willwinder/universalgcodesender/gcode/util/GcodeParserUtils.java#L114) we filter out comments from the commands before saving to the temporary file. 

As far as I can see, our preprocessor for removing comments wouldn't get much to work with?

PR #1364 (2020-03-09): Do not remove comments by default
Fixes #1318

Removed the function in the gcode parsing that removes comments. This should be done by the preprocessor.
closed 2020-03-09 by breiler