winder/Universal-G-Code-Sender #1322: Pendant address error

xunni1000, 2019-12-12:
The pendant address  is 192.168.51.1:8080
It is the IP of virtual network.

My computer address is 192.168.2.180.

```
C:\Users\xjw11>ipconfig

Windows IP 配置

以太网适配器 以太网:

   连接特定的 DNS 后缀 . . . . . . . : lan
   IPv6 地址 . . . . . . . . . . . . : 240e:ce:41e7:4400:48f2:c09e:ae87:5758
   IPv6 地址 . . . . . . . . . . . . : 240e:ce:41e7:4500::180
   IPv6 地址 . . . . . . . . . . . . : 240e:ce:41e7:4500:48f2:c09e:ae87:5758
   IPv6 地址 . . . . . . . . . . . . : 240e:ce:41e7:4600::180
   IPv6 地址 . . . . . . . . . . . . : 240e:ce:41e7:4600:48f2:c09e:ae87:5758
   IPv6 地址 . . . . . . . . . . . . : fdbc:3cd3:f778:0:48f2:c09e:ae87:5758
   临时 IPv6 地址. . . . . . . . . . : 240e:ce:41e7:4400:7cf8:59f0:3dbe:f8d2
   临时 IPv6 地址. . . . . . . . . . : 240e:ce:41e7:4500:7cf8:59f0:3dbe:f8d2
   临时 IPv6 地址. . . . . . . . . . : 240e:ce:41e7:4600:7cf8:59f0:3dbe:f8d2
   临时 IPv6 地址. . . . . . . . . . : fdbc:3cd3:f778:0:7cf8:59f0:3dbe:f8d2
   本地链接 IPv6 地址. . . . . . . . : fe80::48f2:c09e:ae87:5758%8
   IPv4 地址 . . . . . . . . . . . . : 192.168.2.180
   子网掩码  . . . . . . . . . . . . : 255.255.255.0
   默认网关. . . . . . . . . . . . . : fe80::203:7fff:fe11:2001%8
                                       192.168.2.1

无线局域网适配器 WLAN:

   媒体状态  . . . . . . . . . . . . : 媒体已断开连接
   连接特定的 DNS 后缀 . . . . . . . :

无线局域网适配器 本地连接* 1:

   媒体状态  . . . . . . . . . . . . : 媒体已断开连接
   连接特定的 DNS 后缀 . . . . . . . :

无线局域网适配器 本地连接* 2:

   媒体状态  . . . . . . . . . . . . : 媒体已断开连接
   连接特定的 DNS 后缀 . . . . . . . :

以太网适配器 VMware Network Adapter VMnet1:

   连接特定的 DNS 后缀 . . . . . . . :
   本地链接 IPv6 地址. . . . . . . . : fe80::1073:b50a:717e:76d0%6
   IPv4 地址 . . . . . . . . . . . . : 192.168.52.1
   子网掩码  . . . . . . . . . . . . : 255.255.255.0
   默认网关. . . . . . . . . . . . . :

以太网适配器 VMware Network Adapter VMnet8:

   连接特定的 DNS 后缀 . . . . . . . :
   本地链接 IPv6 地址. . . . . . . . : fe80::fcee:d012:d051:3f43%11
   IPv4 地址 . . . . . . . . . . . . : 192.168.131.1
   子网掩码  . . . . . . . . . . . . : 255.255.255.0
   默认网关. . . . . . . . . . . . . :

以太网适配器 蓝牙网络连接:

   媒体状态  . . . . . . . . . . . . : 媒体已断开连接
   连接特定的 DNS 后缀 . . . . . . . :

以太网适配器 vEthernet (Default Switch):

   连接特定的 DNS 后缀 . . . . . . . :
   本地链接 IPv6 地址. . . . . . . . : fe80::38a3:cfd:dc1:834b%25
   IPv4 地址 . . . . . . . . . . . . : 172.17.71.129
   子网掩码  . . . . . . . . . . . . : 255.255.255.240
   默认网关. . . . . . . . . . . . . :

```

breiler, 2019-12-12:
It will listen to all network adapters, so any IP-address should work. 
It's unfortunate that the pendant popup shows the wrong IP, but there is no way to know which address that is the right. 

As I see it we have these options:
* Leave as is
* Extend the popup with all the network adapters available
* Only show the localhost adapter
* Try to figure out which IP that's used to access the Internet and assume that this should be used
* Remove the popup entirely

Any other ideas?

xunni1000, 2019-12-13:
I'm not a java programmer, but I have some ideas
1. The virtual network IP is x.x.x.1 (the last bit of the computer will not be 1). Of course, if the user has modified the last bit, it may not be 1
2. The name of the network adapter is "VMware network adapter VMnet8"
Can we use blacklist to exclude "VMware" string
3. The virtual gateway is empty, but the computer is not empty
4. Or use combobox (HTML tag select / option) to let me choose one

winder, 2019-12-18:
@xunni1000 if you're running UGS inside a virtual machine, you probably need to forward ports to the host computer. You can test that the pendant is running correctly by using a web browser inside the VM.

@breiler the old jetty webserver returned a number of IP addresses if there were more than one. Do you know if that still happens?

breiler, 2019-12-18:
@winder as far as I can see it should still list IP-addresses, but I don't have the possibility to test it myself:

* https://github.com/winder/Universal-G-Code-Sender/blob/master/ugs-pendant/src/main/java/com/willwinder/universalgcodesender/pendantui/PendantUI.java#L111
* https://github.com/winder/Universal-G-Code-Sender/blob/master/ugs-platform/ugs-platform-ugscore/src/main/java/com/willwinder/ugs/nbp/core/actions/PendantAction.java#L77

breiler, 2022-12-16:
Closing as there is not much that we can do from UGS.

closed 2022-12-16 by breiler