winder/Universal-G-Code-Sender #1325: Jog control

chandler99, 2019-12-17:
When using the jog control the motor only move in on direction but when running code it moves in both direction.  Ex when Z jog either positive or negative the both move downwards.

breiler, 2019-12-18:
I'm not sure that I understand. If you click on the jog buttons in the GUI it only moves in the negative direction?

But if you run a gcode program it works fine?

What version of UGS are you running?
What variant of UGS, classic/platform?
What controller and version are you using, eg. grbl 1.1f?

winder, 2019-12-18:
This could also happen if you have a wiring issue

chandler99, 2019-12-18:
As of last night now the g-code program also move in one direction on the x and z axis. I have a grbl .9 controller.  I am using the latest version of UGS platform. I am also very new to this I am just starting to test the machine.

breiler, 2019-12-19:
Now it sounds like a wiring problem that @winder suggested 

Check that you have the direction pins connected correctly:
https://github.com/gnea/grbl/wiki/Connecting-Grbl

closed 2019-12-22 by chandler99