winder/Universal-G-Code-Sender #1326: UGS consistently stops mid-job on RPi/Arduino GRBL setup and goes "Offline" for no clear reason

coogle, 2019-12-19:
## Problem description 

UGS keeps losing it's connection to the controller, causing the job to stop midway through. I am running UGS on a Raspberry Pi 4 (4GB) and my controller is powered by an Arduino UNO running a recent version of GRBL. 

### Expected Behavior

The job should complete. 

### Actual Behavior

Sometimes it goes "offline" 10 minutes in, sometimes 40, other times 15 minutes in. The job also does not show up in the visualizer (might be an indication of something else wrong?) Once it goes offline if I click the connect icon it immediately reconnects and I can in theory start the entire job over again.

The console doesn't show any indication something went wrong, no alerts, etc. it just goes Offline basically without explanation. 

### Steps to Reproduce the Problem

  1. Open the program
  2. Load a Gcode file
  3. Click start
  4. Wait anywhere from 10 to 40 minutes
  5. CNC goes offline.

### Specifications
#### Version

UGS Platform (I'm not at the computer right now but it's a very recent nightly build)
Raspberry Pi 4 / 4GB of ram model
Arduino GRBL powered CNC
Have tested with a different Arduino, and both the official RPI 4 power supply *and* a much beefier PoE Raspberry Pi power adapter

#### Operating system

Raspbian latest, fully updated/upgraded

#### Platform

GRBL 1.1, on an Arduino Uno

#### Other

Things I've tried to address this issue:

- Checked `dmesg` to see if there were errors but nothing stood out
- Tried various USB cables (blue shielded ones) between the RPi and Arduino, as well as various USB ports on the RPi
- Tried the job with and without the router running to eliminate interference
- Swapped out the Arduino for another Arduino (actually swapped the chips between two UNO boards, didn't do a complete reflash yet)

breiler, 2019-12-19:
Have you checked that you are running UGS with the JSSC driver:
https://github.com/winder/Universal-G-Code-Sender/wiki/Raspberry-Pi#change-the-connection-driver

coogle, 2019-12-19:
I did not, I was totally unaware of that page actually. I will review, make all changes and give it another shot and hopefully that solves it!

coogle, 2019-12-19:
Sure enough I wasn't using JSSC so tomorrow when I have time I will run the job that's been failing again with the new serial driver and see if that's the issue and report back.

I know this isn't directly related to the issue but are there updated instructions for the Visualizer / Open GL for the latest Raspberry Pi / Raspbian? The Open GL options have changed (for example only Fake KMS shows up now for me) and it doesn't seem to make the Visualizer work (on an external HDMI screen).

khturnings, 2019-12-19:
Using a Pi3 I found I could get it working by following info in this post - https://github.com/winder/Universal-G-Code-Sender/issues/1247

Last post

winder, 2019-12-19:
I was able to get UGS running on a Pi4 over the weekend, the visualizer worked after switching to the "Fake KMS" driver using the directions linked in the wiki.

breiler, 2020-01-17:
@coogle did you have the chance to run a job with the JSSC driver?

closed 2020-02-28 by breiler