winder/Universal-G-Code-Sender #1327: Loading Gcode : 'Problem with Shortcut'

FloydHD, 2019-12-21:
## Problem description 
When clicking on the open Gcode file it shows me the warning 'Problem with Shortcut'.
After clicking OK, the Messagebox comes up 3 more times and then the File Explorer with the Desktop to select the Gcode File. After the File Explorer opened the Error comes up one more time.

### Expected Behavior
When I try to open a Gcode the File Explorer should show without an error.

### Actual Behavior
3x Error 'Problem with Shortcut'
File Explorer opening
1x Error 'Problem with Shortcut'

### Steps to Reproduce the Problem
Click 'Open Gcode File'

### Specifications
#### Version
UGS Platform (Version 2.0 [nightly] / Aug 20, 2019) 

Product Version: Universal Gcode Platform 20190820
Java: 1.8.0_201; Java HotSpot(TM) 64-Bit Server VM 25.201-b09
Runtime: Java(TM) SE Runtime Environment 1.8.0_201-b09
System: Windows 10 version 10.0 running on amd64; Cp1252; en_US (ugsplatform)
User directory: C:\Users\floyd\.ugsplatform\2.0-SNAPSHOT\dev
Cache directory: C:\Users\floyd\.ugsplatform\2.0-SNAPSHOT\dev\var\cache

#### Operating system
Windows 10

#### Other
[image]

breiler, 2019-12-27:
And you are sure you are opening the actual gcode file and not, as the error suggests, a shortcut-file?

FloydHD, 2019-12-30:
The error occurs before I can even select a Gcode file. 
I've recorded a video : https://youtu.be/0fU1E568ASw

winder, 2019-12-30:
Looks like this is a known issue in the Java runtime. You can fix the desktop shortcut or upgrade to the latest Java 12 version for a fix.

https://bugs.openjdk.java.net/browse/JDK-8213583

FloydHD, 2019-12-30:
So I've currently got Java Version 8 build 1.8.0_231-b11
The Oracle Support for Java 12 has ended and there is only the 13.0.1 Version available. Should I install Version 13?

FloydHD, 2019-12-30:
Ah nvmd. I found the Shortcut which is not working. Deleted it and now it's working perfectly 👍 
Thanks :)

closed 2019-12-30 by FloydHD