winder/Universal-G-Code-Sender #1328: “libjnidispatch-440.jnilib” cannot be opened because the developer cannot be verified.

MrSage, 2019-12-21:
## Problem description 
The latest version of OS X (Catalina) blocks this library from being accessed because it is not properly signed. Gatekeeper warns that `“libjnidispatch-440.jnilib” cannot be opened because the developer cannot be verified.` when opening the ugsplatform script.

I _think_ that this is causing some downstream issues with connecting to my machine but it is hard to say since I wasn't able to find a clear description of what the library does.

### Expected Behavior
Opening the ugsplatform application does not upset the built in gatekeeper.

### Actual Behavior
[image]

### Steps to Reproduce the Problem
  1. Open the ugsplatform on OS X Catalina

### Specifications
#### Version
  UGS Platform 2.0 - Nightly Build Dec 18, 2019

#### Operating system
  Mac OSX Catalina

#### Other
There is also this but I suspect these warnings are a separate thing:
`WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by org.netbeans.ProxyURLStreamHandlerFactory (file:/Users/collinsage/Downloads/ugsplatform/platform/lib/boot.jar) to field java.net.URL.handler
WARNING: Please consider reporting this to the maintainers of org.netbeans.ProxyURLStreamHandlerFactory
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release`

winder, 2019-12-22:
There are several workarounds listed here:
https://github.com/hashicorp/terraform/issues/23033

I'm not sure UGS will ever be a [notarized macos application](https://developer.apple.com/news/?id=09032019a), so one of these workarounds may be the only option.

Could you let me know what works for you? I could update the download page with a comment about the mac os gatekeeper.

MrSage, 2019-12-24:
https://github.com/hashicorp/terraform/issues/23033#issuecomment-542302933

This comment details how to allow the individual library without any drastic security risks.

breiler, 2020-12-08:
Closed as it has a work around. Not much we can do in UGS.

closed 2020-12-08 by breiler