winder/Universal-G-Code-Sender #1329: UGS not working on my macos10.15

planform, 2019-12-22:
## Problem description 
2019-12-22 16:04:51.605 java[10928:1200605] Apple AWT Internal Exception: NSWindow drag regions should only be invalidated on the Main Thread!
2019-12-22 16:04:51.605 java[10928:1200605] *** Terminating app due to uncaught exception 'NSInternalInconsistencyException', reason: 'NSWindow drag regions should only be invalidated on the Main Thread!'
*** First throw call stack:
(
	0   CoreFoundation                      0x00007fff2d9428ab __exceptionPreprocess + 250
	1   libobjc.A.dylib                     0x00007fff63a1a805 objc_exception_throw + 48
	2   CoreFoundation                      0x00007fff2d96b30c -[NSException raise] + 9
	3   AppKit                              0x00007fff2aabb538 -[NSWindow(NSWindow_Theme) _postWindowNeedsToResetDragMarginsUnlessPostingDisabled] + 310
	4   AppKit                              0x00007fff2aaa2ed5 -[NSWindow _initContent:styleMask:backing:defer:contentView:] + 1416
	5   AppKit                              0x00007fff2aaa2947 -[NSWindow initWithContentRect:styleMask:backing:defer:] + 42
	6   libnativewindow_macosx.jnilib       0x000000012a7e73fe Java_jogamp_nativewindow_macosx_OSXUtil_CreateNSWindow0 + 398
	7   ???                                 0x0000000112ef9407 0x0 + 4612658183
)
libc++abi.dylib: terminating with uncaught exception of type NSException
/Users/steve/Downloads/ugsplatform/bin/../platform/lib/nbexec: line 425: 10928 Abort trap: 6           "/Library/Java/JavaVirtualMachines/liberica-jdk-1.8.0_222/Contents/Home/bin/java" -Djdk.home="/Library/Java/JavaVirtualMachines/liberica-jdk-1.8.0_222/Contents/Home" -classpath "/Users/steve/Downloads/ugsplatform/platform/lib/boot.jar:/Users/steve/Downloads/ugsplatform/platform/lib/org-openide-modules.jar:/Users/steve/Downloads/ugsplatform/platform/lib/org-openide-util-lookup.jar:/Users/steve/Downloads/ugsplatform/platform/lib/org-openide-util-ui.jar:/Users/steve/Downloads/ugsplatform/platform/lib/org-openide-util.jar:/Library/Java/JavaVirtualMachines/liberica-jdk-1.8.0_222/Contents/Home/lib/dt.jar:/Library/Java/JavaVirtualMachines/liberica-jdk-1.8.0_222/Contents/Home/lib/tools.jar" -Dnetbeans.dirs="/Users/steve/Downloads/ugsplatform/etc:/Users/steve/Downloads/ugsplatform/ide:/Users/steve/Downloads/ugsplatform/bin:/Users/steve/Downloads/ugsplatform/ugsplatform:/Users/steve/Downloads/ugsplatform/extra:/Users/steve/Downloads/ugsplatform/java:/Users/steve/Downloads/ugsplatform/platform:" -Dnetbeans.home="/Users/steve/Downloads/ugsplatform/platform" '-Xdock:name=ugsplatform' '-Xdock:icon=/Users/steve/Downloads/ugsplatform/bin/../../ugsplatform.icns' '-Xms64m' '-Xverify:none' '-Dsun.java2d.noddraw=true' '-Dsun.awt.noerasebackground=true' '-Dnetbeans.indexing.noFileRefresh=true' -XX:+HeapDumpOnOutOfMemoryError -XX:HeapDumpPath="/Users/steve/Library/Application Support/ugsplatform/2.0-SNAPSHOT/dev/var/log/heapdump.hprof" org.netbeans.Main --cachedir "/Users/steve/Library/Application Support/ugsplatform/2.0-SNAPSHOT/dev/var/cache" --userdir "/Users/steve/Library/Application Support/ugsplatform/2.0-SNAPSHOT/dev" "--branding" "ugsplatform" 0<&0

breiler, 2019-12-27:
Never heard of [Liberica JDK](https://bell-sw.com/) which you seem to have installed.

If you get chance, could you give the build with a bundled Java ([Adopt OpenJDK](https://adoptopenjdk.net/)) a try?
https://ugs.jfrog.io/ugs/webapp/#/artifacts/browse/tree/General/UGS/nightly/ugs-platform-app-2.0-SNAPSHOT.dmg

breiler, 2020-01-10:
Closing as no more information or status was provided.

closed 2020-01-10 by breiler

DanielWeigl, 2020-01-18:
Just FYI, I get the same error on MacOs `macOS 10.14.6` with OpenJDK13 `OpenJDK Runtime Environment (build 13.0.1+9)`

Its seems related to the JOGL lib: https://jogamp.org/bugzilla/show_bug.cgi?id=1370
The issue is reported as RESOLVED, but JOGL didnt have a new release since 2015...

But running it with Java12 (same OS), as you suggested here https://github.com/winder/Universal-G-Code-Sender/issues/1330 works

soukhinov, 2020-09-12:
Yesterday I updated Java on MacOS 10.15.6.

UGSPlatform does not work anymore. Any ideas?

`2020-09-12 16:26:57.343 java[65619:1667091] Apple AWT Internal Exception: NSWindow drag regions should only be invalidated on the Main Thread!
2020-09-12 16:26:57.343 java[65619:1667091] *** Terminating app due to uncaught exception 'NSInternalInconsistencyException', reason: 'NSWindow drag regions should only be invalidated on the Main Thread!'
*** First throw call stack:
(
	0   CoreFoundation                      0x00007fff32254b57 __exceptionPreprocess + 250
	1   libobjc.A.dylib                     0x00007fff6b09b5bf objc_exception_throw + 48
	2   CoreFoundation                      0x00007fff3227d34c -[NSException raise] + 9
	3   AppKit                              0x00007fff2f4775ec -[NSWindow(NSWindow_Theme) _postWindowNeedsToResetDragMarginsUnlessPostingDisabled] + 310
	4   AppKit                              0x00007fff2f45f052 -[NSWindow _initContent:styleMask:backing:defer:contentView:] + 1416
	5   AppKit                              0x00007fff2f45eac3 -[NSWindow initWithContentRect:styleMask:backing:defer:] + 42
	6   libnativewindow_macosx.jnilib       0x000000016171c3fe Java_jogamp_nativewindow_macosx_OSXUtil_CreateNSWindow0 + 398
	7   ???                                 0x00000001156be6c7 0x0 + 4654360263
)
libc++abi.dylib: terminating with uncaught exception of type NSException
/Users/anton/Desktop/CNC/ugsplatform/bin/../platform/lib/nbexec: line 425: 65619 Abort trap: 6           "/Library/Internet Plug-Ins/JavaAppletPlugin.plugin/Contents/Home/bin/java" -Djdk.home="/Library/Internet Plug-Ins/JavaAppletPlugin.plugin/Contents/Home" -classpath "/Users/anton/Desktop/CNC/ugsplatform/platform/lib/boot.jar:/Users/anton/Desktop/CNC/ugsplatform/platform/lib/org-openide-modules.jar:/Users/anton/Desktop/CNC/ugsplatform/platform/lib/org-openide-util-lookup.jar:/Users/anton/Desktop/CNC/ugsplatform/platform/lib/org-openide-util-ui.jar:/Users/anton/Desktop/CNC/ugsplatform/platform/lib/org-openide-util.jar" -Dnetbeans.dirs="/Users/anton/Desktop/CNC/ugsplatform/etc:/Users/anton/Desktop/CNC/ugsplatform/ide:/Users/anton/Desktop/CNC/ugsplatform/bin:/Users/anton/Desktop/CNC/ugsplatform/ugsplatform:/Users/anton/Desktop/CNC/ugsplatform/extra:/Users/anton/Desktop/CNC/ugsplatform/java:/Users/anton/Desktop/CNC/ugsplatform/platform:" -Dnetbeans.home="/Users/anton/Desktop/CNC/ugsplatform/platform" '-Xdock:name=ugsplatform' '-Xdock:icon=/Users/anton/Desktop/CNC/ugsplatform/bin/../../ugsplatform.icns' '-Xms64m' '-Xverify:none' '-Dsun.java2d.noddraw=true' '-Dsun.awt.noerasebackground=true' '-Dnetbeans.indexing.noFileRefresh=true' -XX:+HeapDumpOnOutOfMemoryError -XX:HeapDumpPath="/Users/anton/Library/Application Support/ugsplatform/2.0-SNAPSHOT/dev/var/log/heapdump.hprof" org.netbeans.Main --cachedir "/Users/anton/Library/Application Support/ugsplatform/2.0-SNAPSHOT/dev/var/cache" --userdir "/Users/anton/Library/Application Support/ugsplatform/2.0-SNAPSHOT/dev" "--branding" "ugsplatform" 0<&0`

soukhinov, 2020-09-12:
`java -version
java version "14.0.2" 2020-07-14
Java(TM) SE Runtime Environment (build 14.0.2+12-46)
Java HotSpot(TM) 64-Bit Server VM (build 14.0.2+12-46, mixed mode, sharing)`

soukhinov, 2020-09-13:
Java SDK 13 also gives the same error.

Java SKD 12 works.