winder/Universal-G-Code-Sender #1330: Visualize closes the window

justinefortier, 2019-12-26:
Hello, I have an issue with the visualize. 

When I click on visualize, the software closes. 
I use UGS 2.0 Classic on Mac OS X Catalina and I run it with the command line `java -jar -Xmx256m UniversalGcodeSender.jar`

I use this file : https://github.com/winder/Universal-G-Code-Sender/blob/master/test_files/square.nc
The error appears with any file. I also tried with the previous version of the software, but even resulted.

Here is the message that appears in the terminal :
```
INFO: Log location: /Users/justine/Library/Preferences/ugs/UniversalGcodeSender.json
déc. 26, 2019 2:52:02 PM com.willwinder.universalgcodesender.utils.SettingsFactory loadSettings
INFO: Loading settings.
Dec 26, 2019 2:52:02 PM com.willwinder.universalgcodesender.model.GUIBackend applySettings
INFO: Applying settings.
Dec 26, 2019 2:52:07 PM com.willwinder.universalgcodesender.model.GUIBackend addUGSEventListener
INFO: Adding UGSEvent listener: JogPanel
Dec 26, 2019 2:52:07 PM com.willwinder.universalgcodesender.model.GUIBackend addControllerListener
INFO: Adding controller listener: JogPanel
Dec 26, 2019 2:52:07 PM com.willwinder.universalgcodesender.model.GUIBackend addUGSEventListener
INFO: Adding UGSEvent listener: MacroActionPanel
Dec 26, 2019 2:52:07 PM com.willwinder.universalgcodesender.model.GUIBackend addUGSEventListener
INFO: Adding UGSEvent listener: CommandPanel
Dec 26, 2019 2:52:07 PM com.willwinder.universalgcodesender.model.GUIBackend addUGSEventListener
INFO: Adding UGSEvent listener: CommandTextArea
Dec 26, 2019 2:52:07 PM com.willwinder.universalgcodesender.model.GUIBackend addUGSEventListener
INFO: Adding UGSEvent listener: ConfigureFirmwareAction$$Lambda$186/0x0000000800c63840
Dec 26, 2019 2:52:08 PM com.willwinder.universalgcodesender.model.GUIBackend applySettings
INFO: Applying settings.
Dec 26, 2019 2:52:08 PM com.willwinder.universalgcodesender.utils.FirmwareUtils initialize
INFO: Initializing firmware... ...
Dec 26, 2019 2:52:08 PM com.willwinder.universalgcodesender.utils.FirmwareUtils initialize
INFO: /resources/firmware_config
Dec 26, 2019 2:52:08 PM com.willwinder.universalgcodesender.utils.FirmwareUtils initialize
INFO: /resources/firmware_config/g2core.json
Dec 26, 2019 2:52:08 PM com.willwinder.universalgcodesender.utils.FirmwareUtils initialize
INFO: /resources/firmware_config/test_1.json
Dec 26, 2019 2:52:08 PM com.willwinder.universalgcodesender.utils.FirmwareUtils initialize
INFO: /resources/firmware_config/test_2.json
Dec 26, 2019 2:52:08 PM com.willwinder.universalgcodesender.utils.FirmwareUtils initialize
INFO: /resources/firmware_config/tinyg.json
Dec 26, 2019 2:52:08 PM com.willwinder.universalgcodesender.utils.FirmwareUtils initialize
INFO: /resources/firmware_config/grbl.json
Dec 26, 2019 2:52:08 PM com.willwinder.universalgcodesender.utils.FirmwareUtils initialize
INFO: /resources/firmware_config/smoothieboard.json
Dec 26, 2019 2:52:09 PM com.willwinder.universalgcodesender.model.GUIBackend addControllerListener
INFO: Adding controller listener: MainWindow
Dec 26, 2019 2:52:09 PM com.willwinder.universalgcodesender.model.GUIBackend addUGSEventListener
INFO: Adding UGSEvent listener: MainWindow
Dec 26, 2019 2:52:20 PM com.willwinder.universalgcodesender.model.GUIBackend connect
INFO: Connecting to GRBL on port cu.usbserial-1420
Dec 26, 2019 2:52:20 PM com.willwinder.universalgcodesender.model.GUIBackend applySettings
INFO: Applying settings.
Dec 26, 2019 2:52:20 PM com.willwinder.universalgcodesender.AbstractCommunicator connect
INFO: Connecting to controller using class: JSerialCommConnection with url jserialcomm://cu.usbserial-1420:115200
Dec 26, 2019 2:52:28 PM com.willwinder.universalgcodesender.model.GUIBackend setGcodeFile
INFO: Setting gcode file.
Dec 26, 2019 2:52:28 PM com.willwinder.universalgcodesender.model.GUIBackend initializeProcessedLines
INFO: Start preprocessing
Dec 26, 2019 2:52:28 PM com.willwinder.universalgcodesender.model.GUIBackend preprocessAndExportToFile
INFO: Preprocessing /Users/justine/Desktop/square.nc to /private/var/folders/5p/d6v3025n69v_fs_644b7rnsm0000gn/T/1577368348357-0/square.nc_ugs_1577368348358
Dec 26, 2019 2:52:28 PM com.willwinder.universalgcodesender.model.GUIBackend initializeProcessedLines
INFO: Took 152ms to preprocess
Dec 26, 2019 2:52:28 PM com.willwinder.universalgcodesender.model.GUIBackend getProcessedGcodeFile
INFO: Getting processed gcode file.
Dec 26, 2019 2:52:28 PM com.willwinder.universalgcodesender.utils.SettingsFactory saveSettings
INFO: Saving settings.
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by com.jogamp.common.os.NativeLibrary$3 (file:/Users/justine/Downloads/UniversalGcodeSender/UniversalGcodeSender.jar) to method java.lang.ClassLoader.findLibrary(java.lang.String)
WARNING: Please consider reporting this to the maintainers of com.jogamp.common.os.NativeLibrary$3
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
2019-12-26 14:52:32.818 java[7122:253352] *** Terminating app due to uncaught exception 'NSInternalInconsistencyException', reason: 'NSWindow drag regions should only be invalidated on the Main Thread!'
*** First throw call stack:
(
	0   CoreFoundation                      0x00007fff36ef18ab __exceptionPreprocess + 250
	1   libobjc.A.dylib                     0x00007fff6d162805 objc_exception_throw + 48
	2   CoreFoundation                      0x00007fff36f1a30c -[NSException raise] + 9
	3   AppKit                              0x00007fff3406a538 -[NSWindow(NSWindow_Theme) _postWindowNeedsToResetDragMarginsUnlessPostingDisabled] + 310
	4   AppKit                              0x00007fff34051ed5 -[NSWindow _initContent:styleMask:backing:defer:contentView:] + 1416
	5   AppKit                              0x00007fff34051947 -[NSWindow initWithContentRect:styleMask:backing:defer:] + 42
	6   libnativewindow_macosx.jnilib       0x0000000162cbc3fe Java_jogamp_nativewindow_macosx_OSXUtil_CreateNSWindow0 + 398
	7   ???                                 0x000000011fdc6c90 0x0 + 4829506704
)
libc++abi.dylib: terminating with uncaught exception of type NSException
[1]    7122 abort      java -jar -Xmx256m UniversalGcodeSender.jar
```

Can you help-me?
Thank you!

breiler, 2019-12-27:
What version of Java are you running? Run this in the terminal:
```
java -version
```

justinefortier, 2019-12-27:
Java 13.0.1

breiler, 2019-12-27:
Could you try and downgrade Java to version 12?

justinefortier, 2019-12-31:
It works with Java 12, thank you very much!

closed 2019-12-31 by justinefortier