winder/Universal-G-Code-Sender #1332: UGS Won't Start

motowilliams, 2019-12-27:
On OS Win10 When starting the UGS, the splash screen shows then disappears. This is what the console show while starting from the CLI. Can seem to find where the logs are kept so I don't have much insight yet.

```
The launcher has determined that the parent process has a console and will reuse it for its own console output.
Closing the console will result in termination of the running program.
Use '--console suppress' to suppress console output.
Use '--console new' to create a separate console window.
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by org.netbeans.ProxyURLStreamHandlerFactory (file:/C:/Users/username/Downloads/ugs-platform-app-2.0-SNAPSHOT/ugsplatform/platform/lib/boot.jar) to field java.net.URL.handler
WARNING: Please consider reporting this to the maintainers of org.netbeans.ProxyURLStreamHandlerFactory
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
```

breiler, 2019-12-27:
What version of Java do you have installed? I have had some problems running Java 13, so in that case you could try uninstall Java and install Java 12 instead.

Or you try a UGS build with a bundled with Java here, let us know if you get this to work:
https://ugs.jfrog.io/ugs/webapp/#/artifacts/browse/tree/General/UGS/nightly/ugs-platform-app-2.0-SNAPSHOT-win.zip

breiler, 2020-03-09:
Reopen if you still have issues with this.

closed 2020-03-09 by breiler