winder/Universal-G-Code-Sender #1335: Alarm:2 Soft limit alarm

Thulebug, 2019-12-31:
Hi,
I know my CNC works because it runs with Easel. When I try to use UGS and Vcarve Desktop it gives me the error in the picture. It doesn't matter where on the CNC I place my material. I've rebooted everything multiple times and recreated the g-code file multiple times.

[image]

Thank You

JacksCrafts, 2019-12-31:
What are your firmware settings? Are they set to inches or mm? Bed size?

Thulebug, 2019-12-31:
Hello, not sure about firmware. Set to inches and it's a Bob's E4 24x24 inch bed.

Sent from AT&T Yahoo Mail on Android

breiler, 2019-12-31:
Can you post your settings from the command:
```$$```

Thulebug, 2020-01-01:
Hello, I'm sorry but I don't understand what you want me to do.

JacksCrafts, 2020-01-01:
When you connect you will see your settings.
$0 = 10    (Step pulse time, microseconds)
$1 = 255    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 0    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 0    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 1    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 1    (Report in inches, boolean)
$20 = 0    (Soft limits enable, boolean)
$21 = 0    (Hard limits enable, boolean)

JacksCrafts, 2020-01-01:
You can also see them when you are connected, by going up to the top menu next to File is Machine.
Toward the bottom of the dropdown is firmware settings.

breiler, 2020-01-03:
@Thulebug did you find the output from your settings?

Thulebug, 2020-01-04:
I will be able to look at it Saturday.

breiler, 2020-03-09:
Closing as we didn't get any further response, please reopen again if you still have a problem

closed 2020-03-09 by breiler