winder/Universal-G-Code-Sender #1337: Unable to open UGSplatform on Windows 7 or Windows 10

artrun, 2020-01-02:
Hello,

When i launch UGS the application loads then gets stuck at the following screen:
[image]

The application then freezes and I have to close via task manager.
At first, I thought it was an issue with java but I download UGS with a specific Java version preconfigured but the same issue happened.
https://ugs.jfrog.io/ugs/webapp/#/artifacts/browse/tree/General/UGS/nightly/ugs-platform-app-2.0-SNAPSHOT-win.zip

Here are the logs:
-------------------------------------------------------------------------------
>Log Session: Thursday, January 2, 2020 at 3:04:03 PM Pacific Standard Time
>System Info: 
  Product Version         = Universal Gcode Platform incubator-netbeans-release-404-on-20190319
  Operating System        = Windows 10 version 10.0 running on amd64
  Java; VM; Vendor        = 12.0.2; OpenJDK 64-Bit Server VM 12.0.2+10; AdoptOpenJDK
  Runtime                 = OpenJDK Runtime Environment 12.0.2+10
  Java Home               = C:\Users\aarzunyan\Downloads\ugsplatform-win\jdk\jdk-12.0.2+10-jre
  System Locale; Encoding = en_US (ugsplatform); Cp1252
  Home Directory          = C:\Users\aarzunyan
  Current Directory       = C:\Users\aarzunyan\Downloads\ugsplatform-win
  User Directory          = C:\Users\aarzunyan\.ugsplatform\2.0-SNAPSHOT\dev
  Cache Directory         = C:\Users\aarzunyan\.ugsplatform\2.0-SNAPSHOT\dev\var\cache
  Installation            = C:\Users\aarzunyan\Downloads\ugsplatform-win\etc
                            C:\Users\aarzunyan\Downloads\ugsplatform-win\ide
                            C:\Users\aarzunyan\Downloads\ugsplatform-win\bin
                            C:\Users\aarzunyan\Downloads\ugsplatform-win\ugsplatform
                            C:\Users\aarzunyan\Downloads\ugsplatform-win\extra
                            C:\Users\aarzunyan\Downloads\ugsplatform-win\java
                            C:\Users\aarzunyan\Downloads\ugsplatform-win\platform
                            C:\Users\aarzunyan\Downloads\ugsplatform-win\platform
  Boot & Ext. Classpath   = 
  Application Classpath   = C:\Users\aarzunyan\Downloads\ugsplatform-win\platform\lib\boot.jar;C:\Users\aarzunyan\Downloads\ugsplatform-win\platform\lib\org-openide-modules.jar;C:\Users\aarzunyan\Downloads\ugsplatform-win\platform\lib\org-openide-util-lookup.jar;C:\Users\aarzunyan\Downloads\ugsplatform-win\platform\lib\org-openide-util-ui.jar;C:\Users\aarzunyan\Downloads\ugsplatform-win\platform\lib\org-openide-util.jar
  Startup Classpath       = C:\Users\aarzunyan\Downloads\ugsplatform-win\platform\core\asm-all-5.0.1.jar;C:\Users\aarzunyan\Downloads\ugsplatform-win\platform\core\core-base.jar;C:\Users\aarzunyan\Downloads\ugsplatform-win\platform\core\core.jar;C:\Users\aarzunyan\Downloads\ugsplatform-win\platform\core\org-netbeans-libs-asm.jar;C:\Users\aarzunyan\Downloads\ugsplatform-win\platform\core\org-openide-filesystems-compat8.jar;C:\Users\aarzunyan\Downloads\ugsplatform-win\platform\core\org-openide-filesystems.jar;C:\Users\aarzunyan\Downloads\ugsplatform-win\ugsplatform\core\locale\core_ugsplatform.jar
-------------------------------------------------------------------------------
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Applying settings.
INFO [com.willwinder.ugs.nbp.core.lifecycle.startup]: Loading LocalizingService...
INFO [com.willwinder.ugs.nbp.core.lifecycle.startup]: Loading JogService...
INFO [org.netbeans.core.startup.NbEvents]: Turning on modules:
	org.openide.util.lookup [8.37.1 incubator-netbeans-release-404-on-20190319]
	org.openide.util [9.11.1 incubator-netbeans-release-404-on-20190319]
	org.openide.util.ui [9.12.1 incubator-netbeans-release-404-on-20190319]
	org.openide.modules [7.52.1 incubator-netbeans-release-404-on-20190319]
	org.netbeans.api.annotations.common/1 [1.31.1 incubator-netbeans-release-404-on-20190319]
	org.openide.filesystems [9.14.1 incubator-netbeans-release-404-on-20190319]
	org.openide.awt [7.72.1 incubator-netbeans-release-404-on-20190319]
	org.netbeans.api.progress/1 [1.51.1 incubator-netbeans-release-404-on-20190319]
	org.netbeans.api.progress.nb [1.51.1 incubator-netbeans-release-404-on-20190319]
	org.openide.dialogs [7.46.1 incubator-netbeans-release-404-on-20190319]
	org.openide.nodes [7.49.1 incubator-netbeans-release-404-on-20190319]
	org.openide.windows [6.80.1 incubator-netbeans-release-404-on-20190319]
	org.netbeans.modules.editor.mimelookup/1 [1.44.1 incubator-netbeans-release-404-on-20190319]
	org.openide.text [6.71.1 incubator-netbeans-release-404-on-20190319]
	org.netbeans.api.scripting [1.0.1 incubator-netbeans-release-404-on-20190319]
	org.netbeans.swing.tabcontrol [1.60.1 incubator-netbeans-release-404-on-20190319]
	org.netbeans.swing.outline [1.38.1 incubator-netbeans-release-404-on-20190319]
	org.openide.explorer [6.66.1 incubator-netbeans-release-404-on-20190319]
	org.openide.actions [6.43.1 incubator-netbeans-release-404-on-20190319]
	org.netbeans.modules.queries/1 [1.47.1 incubator-netbeans-release-404-on-20190319]
	org.netbeans.api.templates [1.12.1 incubator-netbeans-release-404-on-20190319]
	org.openide.filesystems.nb [9.14.1 incubator-netbeans-release-404-on-20190319]
	org.openide.loaders [7.71.1 incubator-netbeans-release-404-on-20190319]
	org.netbeans.api.intent [1.6.1 incubator-netbeans-release-404-on-20190319]
	org.netbeans.api.io [1.7.1 incubator-netbeans-release-404-on-20190319]
	org.openide.io [1.53.1 incubator-netbeans-release-404-on-20190319]
	org.openide.filesystems.compat8 [9.13.1 incubator-netbeans-release-404-on-20190319]
	org.netbeans.swing.plaf [1.45.1 incubator-netbeans-release-404-on-20190319]
	org.netbeans.spi.tasklist/1 [1.39.1.1 1 incubator-netbeans-release-404-on-20190319]
	org.netbeans.spi.quicksearch [1.31.1 incubator-netbeans-release-404-on-20190319]
	org.netbeans.spi.navigator/1 [1.42.1 incubator-netbeans-release-404-on-20190319]
[... 121 lines cut ...]
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Adding UGSEvent listener: PortAction
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Adding UGSEvent listener: BaudRateAction
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Adding UGSEvent listener: StartAction
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Adding UGSEvent listener: StopAction
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Adding UGSEvent listener: PauseAction
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Adding UGSEvent listener: HomingAction
INFO [org.netbeans.ui.metrics.laf]: USG_LOOK_AND_FEEL
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Adding UGSEvent listener: LoadedFileStatusLineService$$Lambda$263/0x00000008012b2c40
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Adding UGSEvent listener: JogStatusLine
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Adding UGSEvent listener: SendStatusLine
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Adding controller listener: SendStatusLine
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Getting processed gcode file.
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Adding UGSEvent listener: SendStatusPanel
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Adding controller listener: SendStatusPanel
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Getting processed gcode file.
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Adding UGSEvent listener: MacroActionPanel
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Adding UGSEvent listener: UseSeparateStepSizeAction$$Lambda$271/0x00000008012ca040
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Adding UGSEvent listener: JogTopComponent
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Adding controller listener: ContinuousJogHandler
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Adding UGSEvent listener: ActionButtonPanel
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Adding UGSEvent listener: StateTopComponent
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Adding UGSEvent listener: MachineStatusPanel
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Adding controller state listener: MachineStatusPanel
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Adding UGSEvent listener: CommandPanel
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Adding UGSEvent listener: CommandTextArea

breiler, 2020-01-15:
Thanks for the detailed report!

The latest version of UGS is using a newer version of both the netbeans platform which UGS is built upon and Java version. I'd appreciate if you could give it a try.

The log file content that you posted, was it the complete file? I was expecting some kind of error at the end, but there is no clues... =/

breiler, 2020-01-16:
Similar issue was reported here #1347 caused by the antivirus software.

artrun, 2020-01-17:
thank you. it was the antivirus

breiler, 2020-01-17:
What antivirus software are you using?

closed 2020-01-18 by breiler