winder/Universal-G-Code-Sender #1340: An error was detected while sending '$H': (ALARM:9) Homing fail... on a laser engraver

xeno108, 2020-01-08:
"An error was detected while sending '$H': (ALARM:9) Homing fail. Could not find limit switch within search distances. Try increasing max travel, decreasing pull-off distance, or check wiring. Streaming has been paused."

I purchased a DIY laser engraver from the internets. I pulled out the ardiuno nano and put in my own. I wanted to customize the setup without ruining the original controller.

I want to be able to add limit switches to the project (later project if possible would be to add a safety switch for the lid of my enclosure).

When I connect, I can jog in x and y. I can activate laser as well. Status is "Ready" otherwise. When I $H, status becomes "busy" and the motors lock up. There is no movement.

Because it's a laser, I am not homing Z, only X and Y.

My parameters are as follows:
$$
$0=10
$1=25
$2=0
$3=0
$4=0
$5=0
$6=0
$10=1
$11=1.000
$12=0.002
$13=0
$20=0
$21=0
$22=1
$23=0
$24=25.000
$25=500.000
$26=250
$27=1.000
$30=1000
$31=0
$32=0
$100=80.000
$101=80.000
$102=250.000
$110=8000.000
$111=8000.000
$112=500.000
$120=500.000
$121=500.000
$122=10.000
$130=500.000
$131=500.000
$132=200.000
ok

And my config.h file is below.

I have a Cronos board that houses both the nano and motor controllers (2). Looks similar to the eleksmaker.

Anybody have a clue as to why it's not running? I have searched around and found similar things for CNC with z, but not with laser.

Thanks.

/*
  config.h - compile time configuration
  Part of Grbl

  Copyright (c) 2012-2015 Sungeun K. Jeon
  Copyright (c) 2009-2011 Simen Svale Skogsrud

  Grbl is free software: you can redistribute it and/or modify
  it under the terms of the GNU General Public License as published by
  the Free Software Foundation, either version 3 of the License, or
  (at your option) any later version.

  Grbl is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
[... 382 lines cut ...]
// misread the pin. When hard limits are triggered, they should be 100% reliable, which is the
// reason that this option is disabled by default. Only if your system/electronics can guarantee
// that the switches don't bounce, we recommend enabling this option. This will help prevent
// triggering a hard limit when the machine disengages from the switch.
// NOTE: This option has no effect if SOFTWARE_DEBOUNCE is enabled.
// #define HARD_LIMIT_FORCE_STATE_CHECK // Default disabled. Uncomment to enable.

// ---------------------------------------------------------------------------------------
// COMPILE-TIME ERROR CHECKING OF DEFINE VALUES:

#ifndef HOMING_CYCLE_0
  #error "Required HOMING_CYCLE_0 not defined."
#endif

#if defined(USE_SPINDLE_DIR_AS_ENABLE_PIN) && !defined(VARIABLE_SPINDLE)
  #error "USE_SPINDLE_DIR_AS_ENABLE_PIN may only be used with VARIABLE_SPINDLE enabled"
#endif

#if defined(USE_SPINDLE_DIR_AS_ENABLE_PIN) && !defined(CPU_MAP_ATMEGA328P)
  #error "USE_SPINDLE_DIR_AS_ENABLE_PIN may only be used with a 328p processor"
#endif

// ---------------------------------------------------------------------------------------

#endif

breiler, 2020-01-09:
Your max speed settings are quite high:
$110=8000.000
$111=8000.000
$112=500.000

Is your machine able to move 8000 mm/min, ex. can you jog in these speeds?

xeno108, 2020-01-09:
I can jog at specified speeds. I haven't tried to run it at those speeds. These were the predefined values the board came with.

breiler, 2023-08-21:
I am closing this as it is an issue with GRBL and not UGS.

The config looks fine and should have worked to configure two axis homing instead of three:
```
#define HOMING_CYCLE_0 (1<<X_AXIS)
#define HOMING_CYCLE_1 (1<<Y_AXIS)
```

And the $$-settings looks fine. If you ever found a solution to this I would really like to read about it.

closed 2023-08-21 by breiler