winder/Universal-G-Code-Sender #1344: DRO changes to WCS offset ignore differences between DRO and machine units

dvsingletary, 2020-01-12:
## Problem description 
If Jog/DRO are showing units in inches, but machine state is in MM (G21), typing a raw value into any of the WCS values in the DRO will be sent to the controller without being converted into the units that the controller is currently using.  Same applies vice-versa (bug exists if the DRO/Jog units are different than the controller units)

### Expected Behavior
Typing in a value into the DRO should be interpreted in whatever units are active, and if this doesn't match the controller units, it should be converted to controller units when the G10 L20 command is issued.

### Actual Behavior
G10 L20 is sent to the controller using whatever value was typed into the DRO without conversion if needed.

### Steps to Reproduce the Problem
1) Set DRO/Jog units to Inches (using the inch/mm button in the Jog controls)
2) Set controller state to mm (issue G21 command in console)
3) click on any of the WCS coordinate values in the DRO to change it
4) enter a non-zero value
5) observe that the value is sent (via G10 L20) to the controller without converting into the controller's units
6) DRO shows a different value than what was entered, because it's showing the actual coordinate that the controller learned.

I'll take a crack at fixing this.

breiler, 2020-01-12:
I've noticed this a couple of times but wasn't able to reproduce it reliably. This is great if you manage to fix it.

dvsingletary, 2021-02-28:
I saw that there have been some changes to this code since I filed this issue, and I'm no longer to reproduce it (ie bug is fixed).

closed 2021-02-28 by dvsingletary