winder/Universal-G-Code-Sender #1345: Pendant ntroller z step size < 1 is broken

arneboe, 2020-01-12:
### Problem
Unable to enter values < 1 for z step size in web pendant.

### Expected Behavior
I am able to enter 0.1 into the field for z step size.

### Actual Behavior
The "." is removed and the value turns into 1 instead of 0.1.

### Steps to Reproduce the Problem
1. open ugs
2. open the web pendant
3. go to the jog controller
4. try to enter a value < 1 into z step size.

### Specifications
#### Version
  UGS Platform 2.0 - Nightly Build Aug 14 2019 (this is named stable on the download page).
#### Operating system
  Windows 10

breiler, 2020-01-12:
This could be caused by the language specific settings of the web browser. What happens if you write a comma: 0,1?

arneboe, 2020-01-14:
Doesn't work either. The comma is removed immediately

arneboe, 2020-01-14:
I am accessing the pendant using firefox on android if that is relevant

PR #1382 (2020-04-02): Removed step size functionality as it wasn't longer possible to set a value manually on some devices.

Fixes #1345
closed 2020-04-02 by breiler