winder/Universal-G-Code-Sender #1347: UGS platform loads with Nightly build screen and locks up

lappa1, 2020-01-15:
## Feature request

## Problem description 

### Expected Behavior

### Actual Behavior

### Steps to Reproduce the Problem

  1.
  1.
  1.

### Specifications
#### Version

#### Operating system

#### Platform

#### Other

breiler, 2020-01-15:
You don't give a lot information to work with.

* What operating system are you using?
* Which version have you downloaded
* What Java version are you using?

It would be really helpful if you attached the logs, the locations of the log files are described here:
http://winder.github.io/ugs_website/guide/troubleshooting/#property-files

lappa1, 2020-01-15:
Sorry but I thought that is just for a heading for the post.
I'm using WIN 10
Java is the latest - Ver 8  Update 241
I downloaded the one under "Stable" 2.0 Platform (Beta) Aug 14, 2019
Here's a screen shot. I can't exit the program without using Task manager.
The only section selectable is the Nightly Build screen but I cant close it or select anything else on the screen
[image]

lappa1, 2020-01-15:
here's the log file as requested.
[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/4063366/messages.log)

breiler, 2020-01-15:
Thanks!

This looks like a duplicate of #1337. Is it possible for you to try the latest nightly:
https://ugs.jfrog.io/ugs/UGS/nightly/ugs-platform-app-2.0-SNAPSHOT-win.zip

lappa1, 2020-01-15:
Thanks for the reply. I downloaded and installed but Its exactly the same.
Here's the new logfile if that helps.
[messages (1).log](https://github.com/winder/Universal-G-Code-Sender/files/4063764/messages.1.log)

breiler, 2020-01-15:
No clues from the log. Do you have a antivirus software that could interfere? We've seen this before:
https://github.com/winder/Universal-G-Code-Sender/wiki/Troubleshooting-Overview#antivirus-software

lappa1, 2020-01-15:
I use Trend Micro which has never caused problems before but I'll switch it off and see if it makes a difference. There is no way I would download a new UGS platform with the virus software off though.

lappa1, 2020-01-16:
OK. Turned off Trend Micro and all works fine. What was a blank Nightly build box in the center of the screen now has messages, It can be closed and I can use the program. I am using the Snapshot one you posted the link for,
Is there anything else you would like me to try that may help others?
Thanks so much for your help.

lappa1, 2020-01-16:
BTW. To test things further I turned Trend Micro back on and tried to open the program. The program once again failed to load properly.
I then turned off Trend Micro, loaded the program successfully, then turned Trend Micro back on (with the program loaded)  and the program continued to function normally.

breiler, 2020-01-16:
Thanks for your feedback. I have no idea why the AV-software reacts this way, especially wihout warning the user its reacting to something.

lappa1, 2020-01-16:
I have contacted the AV vendor and await a reply.

closed 2020-01-18 by breiler

lappa1, 2020-01-24:
Follow up:
I had Trend Micro on my computer for 3 hours today to try and find the reason why Platform locks up on loading,  when Trend Micro is running. 
They tried lots of things like modified .dll files, Registry additions etc. etc.
Still haven’t totally fixed the problem but have put in a temporary fix that allows the program to load and work with Trend Micro running.
They took a number of  Log Files during the events, as well as the Platform program and will work on it to find the problem and apply a fix.
I’ll keep you posted.

lappa1, 2020-01-24:
I’m not sure if I can post now the case is Closed. If I get no feedback I will start a new thread so I can hopefully post a fix,