winder/Universal-G-Code-Sender #1348: Visualizer crashes sender

ArthurSchwieger, 2020-01-15:
[hs_err_pid4888.log](https://github.com/winder/Universal-G-Code-Sender/files/4063065/hs_err_pid4888.log)

## Problem description 
I have a new laptop running Windows 10 Pro.  I have updated to the latest version of Java and latest version of UGC-Classic.  When I load g-code then select Visualize, the program crashes.  I have attached the log.

### Expected Behavior
When I click Visualize, a new window opens and I see the cut path and my tool position.

### Actual Behavior
Entire program crashes.

### Steps to Reproduce the Problem
  1. Open program
  1. Click Browse
  1. Select g-code file that I have cut before
  1. Click Visualize

### Specifications
#### Version
Version 2.0 [nightly] /Dec 06, 2018

#### Operating system
  Windows 10 Pro

#### Platform
GRBL 1.1f

#### Other

breiler, 2020-01-15:
The version you are trying is quite old, could you give the latest nightly build a try?

ArthurSchwieger, 2020-01-15:
I am running what is listed as the nightly build under Downloads/UGS Classic on https://github.com/winder/Universal-G-Code-Sender.  Am I in the wrong place?

ArthurSchwieger, 2020-01-15:
I went to http://winder.github.io/ugs_website/download/ and downloaded a newer file.  It is version 2.0 nightly / Jan 06, 2020.  Still having the same issue.  I got my router connected, homed, loaded the g code file, clicked visualize, and it crashed.  :-(
[hs_err_pid11376.log](https://github.com/winder/Universal-G-Code-Sender/files/4068048/hs_err_pid11376.log)

ArthurSchwieger, 2020-01-15:
Solved!  I found the issue.  It was the version of Java that I had.  Java was telling me I had the latest but when I downloaded it from the link next to the nightly build.  Now it is running great!  :-)

closed 2020-01-15 by ArthurSchwieger