winder/Universal-G-Code-Sender #1350: Macro looking for help

MattiaConfalonieri, 2020-01-27:
Hi everybody, I am new to UGS. I am looking for help to create a macro that moves the CNC to TOP-RIGHT (opposite home) and not depend on what coordinate is the machine when I trigger the macro... is thee possible? that anyone helps me to achieve this.

breiler, 2020-01-28:
Hi and welcome! 😄 

I haven't tested this myself, but in theory you should be able to use the following:
`G53 G90 G21 G0 X100 Y100 Z100`

Which means:
`G53` move in machine coordinates
`G90` move in absolute mode to a given coordinate
`G21` move in millimeters
`G0 X100 Y100 Z100` move using rapid movement (your machine's maximum speed) to the given machine coordinates

You might also want to split the movement for Z so that it will clear any material first, then move XY. In macros you can separate commands using semicolon `;`:
`G53 G90 G21 G0 Z100; G0 X100 Y100`

MattiaConfalonieri, 2020-01-29:
Thanks i would try

closed 2020-02-07 by MattiaConfalonieri