winder/Universal-G-Code-Sender #1351: UGS Platform not loading - continuation with possible fix as Case was closed by Joacim Breiler?

lappa1, 2020-01-28:
Follow up:
I had Trend Micro on my computer for 3 hours today to try and find the reason why Platform locks up on loading, when Trend Micro is running.
They tried lots of things like modified .dll files, Registry additions etc. etc.
Still haven’t totally fixed the problem but have put in a temporary fix that allows the program to load and work with Trend Micro running.
They took a number of Log Files during the events, as well as the Platform program and will work on it to find the problem and apply a fix.
I just got an email from them today. I'll post it below

"This is Mikka from Trend Micro Consumer Support.

We already have analyzed the logs that we have collected the last time I connected in your device. And upon checking here in my end the program Universal G-code Sender does not have digital signature coming from the publisher or company, and that's the reason why Trend Micro detects it as a malicious because Trend Micro was not able to recognize the details of the program. 

To further analyze the program we need to collect another logs which will check the running process of the program when Trend Micro is enabled. 

I apologize for the inconvenience that this has been causing you. Our team is more than willing to solve this concern. In connection, please provide the following details for the call back for us to continue the trouble shooting:"

Noting her reply, is there a problem with the digital signature or doesn't UGS Platform have one??

breiler, 2020-01-28:
Thank you for looking into this.

We don't sign the application for any platform (Mac, Linux, Windows) which we probably should, but:
* This comes with an annual fee for both Mac and Windows, money which at least I don't want to pull from my own pocket. ☹️ 
* It requires (or at least it did for Mac signing) a company with a registered DUNS. 
* Is the tools for signing the software in different platforms available on our automatic build platform

If we were a company selling this software it would probably make sense to priority this. But as a loosely connected team of developers scattered all over the world making software in their free time, I rather focus on the features of the software.

Does anyone else have more experience with this?

lappa1, 2020-01-28:
Thanks for the quick reply.
I can see your dilemma and I for one thank you very much for this software 👍 
As you can see from my post, Trend Micro are arranging another online session with me later in the week to solve the problem. Once I have any answers, I'll post them here.

lappa1, 2020-02-21:
Had a final session with Trend Micro tonight. They developed a patch and installed it tonight and it has fixed the problem.
The guy that applied the patch wasn't quite sure what it actually is but the file name seemed to suggest it was an exception list. He is based in the Philippines.
He is going to contact the techs in the USA and email me further information.
So if anyone running Trend Micro has the problem that the UGS Platform wont load and run, Trend now has a fix.
Hope that helps.

lappa1, 2020-02-21:
Joachim,
I have the answer to what in the UGSPlatform program is causing the problem with TrendMico and maybe other Virus programs. 
Can I contact you directly with Trend’s Email?

Peter Wright

breiler, 2020-02-25:
Thank you Peter! Forward the e-mail and I can have a look.

closed 2020-03-20 by breiler