winder/Universal-G-Code-Sender #1352: Universal G-Code Sender running behind cnc machine

TheNetStriker, 2020-01-28:
I'm running UGS on a Raspberry Pi 4 with a Sainsmart 3018 pro. When I run a bigger gcode file I have the problem that the Visualizer and the console output get's behind the actual movements of the CNC machine. When the machine finishes the Visualizer and the console is still running and I cannot abort the process and have to wait until the output in the console finished. (This can take several minutes) The only workarround I found so far is to close the Visualizer before starting the gcode file.

@KEITHABEAR also reported this issue in the grbl project: https://github.com/gnea/grbl/issues/789

Is this a known problem or is there a workarround for this besides simply closing the Visualizer?

breiler, 2020-01-28:
Yes this is a known issue. Would you mind sharing your gcode file for us to test with?

TheNetStriker, 2020-01-28:
@breiler Here is a link to the gcode files: https://www.dropbox.com/sh/jd5mvldl9bcdlyg/AADFC6u4TinjOLLcxEjf1I-ha?dl=0

The first file is small, the problem there is not noticable, but on the second and third file I had the problem.

KEITHABEAR, 2020-02-20:
Does any know how to fix this it started doing it again

breiler, 2020-04-01:
I've made some changes in the visualizer that might fix this. I wasn't able to access your gcode file so I couldn't test. I'm closing this, please reopen if the problem still exist.

closed 2020-04-01 by breiler