winder/Universal-G-Code-Sender #1354: MVN Errors

ws-mkulas, 2020-02-02:
`[INFO]
[INFO] Results:
[INFO]
[ERROR] Failures:
[ERROR] com.willwinder.ApplicationTest.testApplication
[INFO]   Run 1: PASS
[ERROR]   Run 2: ApplicationTest.testApplication NbModuleSuite has been started with failOnMessage(WARNING) and failOnException(INFO). The following failures have been captured:
[com.willwinder.universalgcodesender.utils.Version] THREAD: On Start/Stop MSG: Couldn't convert the lut 02, 2020 from date format "MMM dd, yyyy" to format "yyyyMMdd"

[INFO]
[INFO]
[ERROR] Tests run: 1, Failures: 1, Errors: 0, Skipped: 0
[INFO]
[INFO] ------------------------------------------------------------------------
[INFO] Reactor Summary for ugs-parent 2.0-SNAPSHOT:
[INFO]
[INFO] ugs-parent ......................................... SUCCESS [  0.506 s]
[INFO] ugs-core ........................................... SUCCESS [ 34.307 s]
[INFO] ugs-pendant ........................................ SUCCESS [ 41.652 s]
[INFO] ugs-cli ............................................ SUCCESS [  0.524 s]
[INFO] ugs-classic ........................................ SUCCESS [  5.056 s]
[INFO] ugs-platform-parent ................................ SUCCESS [  0.234 s]
[INFO] ugs-platform-branding .............................. SUCCESS [  1.836 s]
[INFO] ugs-platform-ugslib ................................ SUCCESS [  7.145 s]
[INFO] ugs-platform-ugscore ............................... SUCCESS [  5.887 s]
[INFO] ugs-platform-visualizer ............................ SUCCESS [  5.812 s]
[INFO] ugs-platform-plugin-workflow ....................... SUCCESS [  2.582 s]
[INFO] ugs-platform-gcode-editor .......................... SUCCESS [  4.922 s]
[INFO] ugs-platform-surfacescanner ........................ SUCCESS [  2.642 s]
[INFO] ProbeModule ........................................ SUCCESS [  4.034 s]
[INFO] DowelModule ........................................ SUCCESS [  9.120 s]
[INFO] GcodeTools ......................................... SUCCESS [  2.356 s]
[INFO] ugs-platform-plugin-jog ............................ SUCCESS [  8.120 s]
[INFO] ugs-platform-plugin-setup-wizard ................... SUCCESS [  3.708 s]
[INFO] ugs-platform-welcome-page .......................... SUCCESS [  2.631 s]
[INFO] ugs-platform-app ................................... FAILURE [ 12.786 s]
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time:  02:37 min
[INFO] Finished at: 2020-02-02T23:03:43+01:00
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal org.apache.maven.plugins:maven-surefire-plugin:3.0.0-M4:test (default-test) on project ugs-platform-app: There are test failures.
[ERROR]
[ERROR] Please refer to /Users/mkulas/Downloads/Universal-G-Code-Sender-master/ugs-platform/application/target/surefire-reports for the individual test results.
[ERROR] Please refer to dump files (if any exist) [date].dump, [date]-jvmRun[N].dump and [date].dumpstream.
[ERROR] -> [Help 1]
[ERROR]
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR]
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoFailureException
[ERROR]
[ERROR] After correcting the problems, you can resume the build with the command
[ERROR]   mvn <args> -rf :ugs-platform-app`

breiler, 2020-02-04:
What language is your system set to?

breiler, 2020-02-23:
Closing as there was no more information from the user.

closed 2020-02-23 by breiler