winder/Universal-G-Code-Sender #1356: [Error] An error was detected while sending '$H': ALARM: Homing fail. Streaming has been paused.

gadgetman53, 2020-02-18:
## Feature request

## Problem description 
I am able to jog X, Y, and Z axis using UGS. I have hardwired limit switches which I have tested as well. When attempting to home, I receive the following errors and have yet to find what is actually causing the issue:
>>> $H
[Error] An error was detected while sending '$H': ALARM: Homing fail. Streaming has been paused.
[Error] Error while processing response <ok>: An unexpected command was completed by the controller.
Grbl 0.9j ['$' for help]
>>> $$
$0 = 10    (step pulse, usec)
$1 = 25    (step idle delay, msec)
$2 = 0    (step port invert mask:00000000)
$3 = 0    (dir port invert mask:00000000)
$4 = 0    (step enable invert, bool)
$5 = 0    (limit pins invert, bool)
$6 = 0    (probe pin invert, bool)
$10 = 19    (status report mask:00010011)
$11 = 0.010    (junction deviation, mm)
$12 = 0.002    (arc tolerance, mm)
$13 = 0    (report inches, bool)
$20 = 0    (soft limits, bool)
$21 = 1    (hard limits, bool)
$22 = 1    (homing cycle, bool)
$23 = 1    (homing dir invert mask:00000001)
$24 = 25.000    (homing feed, mm/min)
$25 = 500.000    (homing seek, mm/min)
$26 = 250    (homing debounce, msec)
$27 = 1.000    (homing pull-off, mm)
$100 = 180.000    (x, step/mm)
$101 = 180.000    (y, step/mm)
$102 = 652.000    (z, step/mm)
$110 = 500.000    (x max rate, mm/min)
$111 = 500.000    (y max rate, mm/min)
$112 = 500.000    (z max rate, mm/min)
$120 = 10.000    (x accel, mm/sec^2)
$121 = 10.000    (y accel, mm/sec^2)
$122 = 10.000    (z accel, mm/sec^2)
$130 = 200.000    (x max travel, mm)
$131 = 200.000    (y max travel, mm)
$132 = 200.000    (z max travel, mm)

### Expected Behavior

### Actual Behavior

### Steps to Reproduce the Problem

  1.
  1.
  1.

### Specifications
#### Version

#### Operating system

#### Platform

#### Other

gadgetman53, 2020-02-18:
I have toggled $23 between 0 and 1 with no change in Z axis direction. I thought the axis should begin upward, but it moves downward. When I just use the jogging, all of my axis move in their appropriate direction.

gadgetman53, 2020-02-18:
I'm using an Arduino UNO with CNC shield 3.0 
Product Version: Universal Gcode Platform 20200106
Java: 1.8.0_241; Java HotSpot(TM) 64-Bit Server VM 25.241-b07
Runtime: Java(TM) SE Runtime Environment 1.8.0_241-b07
System: Windows 10 version 10.0 running on amd64; Cp1252; en_US (ugsplatform)

breiler, 2020-02-23:
So the machine starts moving the Z-axis?

Is it supposed to move more than 200mm that is setup in $132? 
This value also act as the maximum seek distance for homing so increase that value to your Z-max distance. 

Regarding the homing direction you should have a look at this table:
https://github.com/gnea/grbl/wiki/Set-up-the-Homing-Cycle#homing-direction

abdset, 2020-03-15:
This may help.
https://github.com/winder/Universal-G-Code-Sender/issues/1208

gadgetman53, 2020-03-17:
Thank you all for the assistance. Based on feedback, I examined the shield as well as the Arduino and there was no visual soldering issues. So I swapped out the Arduino board with a replacement and that corrected the issue. Not sure why pin 12 did not work, but my issue is resolved.

closed 2020-03-17 by gadgetman53

abdset, 2020-03-17:
The following text is token from config.h file of grbl 1.1h and its speaking about how to use pin11 or pin12 to be a Z limit switch or spindle enable:

// Enables variable spindle output voltage for different RPM values. On the Arduino Uno, the spindle
// enable pin will output 5V for maximum RPM with 256 intermediate levels and 0V when disabled.
// NOTE: IMPORTANT for Arduino Unos! When enabled, the Z-limit pin D11 and spindle enable pin D12 switch!
// The hardware PWM output on pin D11 is required for variable spindle output voltages.
//#define VARIABLE_SPINDLE // Default enabled. Comment to disable. // Aby: i disabled this to use regular D11 pin for Z limit switch.