winder/Universal-G-Code-Sender #1359: AutoLeveller Triggers An Alarm That Cannot Be Reset

TekMason, 2020-03-01:
## Problem description 
If the probe hits the Min Z value during an AutoLeveller Surface Scan an Alarm is triggered that cannot be cleared (with Soft Reset and Unlock).  About 1,000 Errors are displayed in the Console display and then finally all Console messages stop and a Java error "java.lang.OutOfMemoryError: unable to create new native thread" is created
Also:
-the Pause, Play and Stop icons are all greyed out
-Jog Controls are unresponsive
-Home and Return to Zero  work

### Expected Behavior
Be able to clear the alarm and resume the probe that it failed on.

### Actual Behavior
See above

### Steps to Reproduce the Problem
1. Start USG Platform
2. Home Machine
3. Jog to and (Re)set Zero
4. Set parameters in AutoLeveller tab (see screen shots)
5. Execute Scan Surface
6. Probe out a few coordinates normally
7. Let one of the probes reach the Min Z

### Specifications
#### Version
UGS Platform 2.0 - Nightly Build Feb 27, 2020

#### Operating system
Windows 10 x86

#### Platform
Grbl 1.1g

#### Other
Java Version 8 Update 241 (build 1.8.0_241-b07)

#### Attachments Description
ConsoleMessages-Line132.txt
Line 132 is where the probe is allowed to go to Z Min.
[ConsoleMessages-Line132.txt](https://github.com/winder/Universal-G-Code-Sender/files/4272643/ConsoleMessages-Line132.txt)

JavaError.txt
A copy of the text from the error that pops up.
[JavaError.txt](https://github.com/winder/Universal-G-Code-Sender/files/4272646/JavaError.txt)

Link to Screen Shots
https://photos.app.goo.gl/JrrjtDyk2rT3EF116

Please let me know if any further information is needed.

TekMason

TekMason, 2020-03-02:
**Since updating to Grbl 1.1h**:
-the alarm can now be reset by just clicking Unlock
-there are no Java Errors

However,
-there are still about 1,000 Errors are displayed in the Console
-there is no way to resume the surface scan with the Pause, Play and Stop icons greyed out

Grbl 1.1h seems to have fixed the bug and this issue has now become a Feature Request.

_Owners of X-Carves that use the Grbl motion controller, make sure you board from the default grbl v1.1g to v1.1h_

TekMason

winder, 2020-03-02:
It's because [this loop doesn't check for an error](https://github.com/winder/Universal-G-Code-Sender/blob/master/ugs-platform/ugs-platform-surfacescanner/src/main/java/com/willwinder/ugs/platform/surfacescanner/AutoLevelerTopComponent.java#L566).

These commands should be streamed in a way similar to [how it is done with the outline feature](https://github.com/winder/Universal-G-Code-Sender/blob/master/ugs-platform/ugs-platform-ugscore/src/main/java/com/willwinder/ugs/nbp/core/actions/OutlineAction.java#L106). Additionally it might be a good idea for the backend commands like **probe** and **sendGcodeCommand** to throw an exception if the machine is in an ALARM state.

TekMason, 2020-03-02:
> Thanks for the report. I'm surprised that 1.1g and 1.1h have different behaviors.

I'm not terribly surprised as there has never been an official release of grbl 1.1g likely because it was buggier than a Louisiana swamp in July.
Inventables released 1.1g several months back as their latest Easel Driver for the X-Carve, maybe it was custom-rolled by them.  Since the Easel software is very simple and limited (no probing or surface scanning) they wouldn't have tested that kind of functionality.

breiler, 2025-04-22:
The error message loop has been fixed in the nightly build

closed 2025-04-22 by breiler