winder/Universal-G-Code-Sender #1362: g2core

crazyxangel, 2020-03-06:
## Problem description 
UGS platform(nightly) refuses to attempt connecting to arduino DUE when g2core firmware is selected if i select grbl firmware for example it does connect(but doesnt work obviously)
it also works fine on UGS classic (nightly) 

### Expected Behavior
for it to connect or even attempt to connect to the DUE

### Actual Behavior
it literally does nothing not even console is blank aswell

### Steps to Reproduce the Problem
  1. Open the program
  2. Select g2core firmware
  3. Press connect

#### Version
  UGS Platform 2.0 - Nightly Build Mar 23, 2018

#### Operating system
  Windows 10
-->

#### Platform
Arduino DUE g2core

crazyxangel, 2020-03-06:
so redownloading ugs platform for a third time it did try to connect?
now all i had to do was keep on pressing the reset button before it would finally connect.
not sure why ugs platform is acting so strangly compared to the to classic version

breiler, 2020-03-06:
Which version of g2core are you using?

crazyxangel, 2020-03-06:
101.02

breiler, 2020-03-07:
I managed to reproduce this, but I couldn't find a pattern for reproducing it reliably.
When it happened, UGS didn't receive the initialization response from g2core, to fix it I needed unplug the controller for a complete reboot and after connecting I received the required boot message:
`{"r":{"fv":0.99,"fb":101.03,"fbs":"101.03-15-g9fd1","fbc":"settings_default.h","hp":"ArduinoDue","hv":"na","id":"0084-e739-29c6-29c","msg":"SYSTEM READY"},"f":[1,0,1]}`

You wrote:
> not sure why ugs platform is acting so strangly compared to the to classic version

It's the same code for connecting to the controller in both classic and platform version so this should not be an issue. 

@justinclift have you guys experienced this while developing on g2core, is there something we could add to UGS to force a reset when connecting? Couldn't find anything done differently in cncjs.

justinclift, 2020-03-07:
Interesting.  The only weird thing that springs to mind that *could* be related somehow is: https://github.com/synthetos/g2/pull/453

That being said, @giseburt has rewritten a substantial amount of the code base (`edge-preview` branch) over the last few months, cleaning up weirdness and bugs on the way through.  We need to adjust some things for that branch to compile on the Due again though (being worked on), which it doesn't atm.

That probably doesn't help for this issue though. :frowning:

breiler, 2020-04-15:
Not much we can do from UGS, it seems to be a hardware issue.

closed 2020-04-15 by breiler

justinclift, 2020-04-15:
On a related data point, the `edge-preview` branch in g2core compiles on the Due again now. :wink: