winder/Universal-G-Code-Sender #1366: Menu modules not loaded

bsproj, 2020-03-11:
Hi,
when I execute UGS Platform, the following menus are not loaded, showing:
<Connection>
<StartPauseStop>
<Machine Actions>
<Pendant>

How can I fix this problem?

Thank you.

breiler, 2020-03-12:
What does your log file say?
http://winder.github.io/ugs_website/guide/troubleshooting/#property-files

Which version are you running?

bsproj, 2020-03-17:
Hi,
I'm using the last version of USG. I'have encountered this issue also in other recent releases. This is the log:

`-------------------------------------------------------------------------------
>Log Session: Wednesday, March 11, 2020 at 8:26:49 PM Central European Standard Time
>System Info: 
  Product Version         = Universal Gcode Platform 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54
  Operating System        = Windows 10 version 10.0 running on amd64
  Java; VM; Vendor        = 13.0.1; OpenJDK 64-Bit Server VM 13.0.1+9; AdoptOpenJDK
  Runtime                 = OpenJDK Runtime Environment 13.0.1+9
  Java Home               = C:\ugsplatform-win\jdk\jdk-13.0.1+9-jre
  System Locale; Encoding = it_IT (ugsplatform); Cp1252
  Home Directory          = C:\Users\nikit
  Current Directory       = C:\ugsplatform-win
  User Directory          = C:\Users\nikit\.ugsplatform\2.0-SNAPSHOT\dev
  Cache Directory         = C:\Users\nikit\.ugsplatform\2.0-SNAPSHOT\dev\var\cache
  Installation            = C:\ugsplatform-win\etc
                            C:\ugsplatform-win\ide
                            C:\ugsplatform-win\bin
                            C:\ugsplatform-win\ugsplatform
                            C:\ugsplatform-win\extra
                            C:\ugsplatform-win\java
                            C:\ugsplatform-win\platform
                            C:\ugsplatform-win\platform
  Boot & Ext. Classpath   = 
  Application Classpath   = C:\ugsplatform-win\platform\lib\boot.jar;C:\ugsplatform-win\platform\lib\org-openide-modules.jar;C:\ugsplatform-win\platform\lib\org-openide-util-lookup.jar;C:\ugsplatform-win\platform\lib\org-openide-util-ui.jar;C:\ugsplatform-win\platform\lib\org-openide-util.jar
  Startup Classpath       = C:\ugsplatform-win\platform\core\asm-all-5.0.1.jar;C:\ugsplatform-win\platform\core\core-base.jar;C:\ugsplatform-win\platform\core\core.jar;C:\ugsplatform-win\platform\core\org-netbeans-libs-asm.jar;C:\ugsplatform-win\platform\core\org-openide-filesystems-compat8.jar;C:\ugsplatform-win\platform\core\org-openide-filesystems.jar;C:\ugsplatform-win\ugsplatform\core\locale\core_ugsplatform.jar
-------------------------------------------------------------------------------
INFO [com.willwinder.universalgcodesender.utils.SettingsFactory]: Log location: C:\Users\nikit\.ugs\UniversalGcodeSender.json
INFO [com.willwinder.universalgcodesender.utils.SettingsFactory]: Loading settings.
INFO [com.willwinder.ugs.nbp.core.lifecycle.startup]: Loading LocalizingService...
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Applying settings.
INFO [com.willwinder.ugs.nbp.core.lifecycle.startup]: Loading JogService...
INFO [org.netbeans.core.startup.NbEvents]: Turning on modules:
	org.openide.util.lookup [8.39 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54]
	org.openide.util [9.13 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54]
	org.openide.util.ui [9.14 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54]
	org.openide.modules [7.54 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54]
	org.netbeans.api.annotations.common/1 [1.33 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54]
	org.openide.filesystems [9.16 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54]
	org.openide.awt [7.74 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54]
	org.netbeans.api.progress/1 [1.53 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54]
	org.netbeans.api.progress.nb [1.53 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54]
	org.openide.dialogs [7.48 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54]
	org.openide.nodes [7.51 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54]
	org.openide.windows [6.83 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54]
	org.netbeans.modules.editor.mimelookup/1 [1.46 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54]
	org.openide.text [6.73 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54]
	org.netbeans.api.scripting [1.3 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54]
	org.netbeans.swing.tabcontrol [1.62 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54]
	org.netbeans.swing.outline [1.40 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54]
	org.openide.explorer [6.68 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54]
	org.openide.actions [6.45 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54]
	org.netbeans.modules.queries/1 [1.49 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54]
	org.netbeans.api.templates [1.14 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54]
	org.openide.filesystems.nb [9.16 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54]
	org.openide.loaders [7.73 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54]
	org.netbeans.api.intent [1.8 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54]
	org.netbeans.api.io [1.9 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54]
	org.openide.io [1.55 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54]
	org.openide.filesystems.compat8 [9.15 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54]
	org.netbeans.swing.plaf [1.47 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54]
	org.netbeans.spi.tasklist/1 [1.41.0.1 1 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54]
	org.netbeans.spi.quicksearch [1.33 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54]
	org.netbeans.spi.navigator/1 [1.44 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54]
	org.netbeans.modules.editor.settings/1 [1.63 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54]
	org.netbeans.modules.editor.util/1 [1.71 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54]
	org.netbeans.modules.editor.document [1.15.0.3 3 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54]
[... cut ...]
INFO [com.willwinder.ugs.nbm.visualizer.shared.GcodeRenderer]: Disposing OpenGL context.
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Adding UGSEvent listener: OverridesPanel
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Adding controller listener: OverridesPanel
INFO [com.willwinder.universalgcodesender.utils.SettingsFactory]: Saving settings.
INFO [org.netbeans.ui.indexing]: Indexing started, time from last indexing 0 ms.
INFO [org.netbeans.ui.indexing]: Indexing finished, indexing took 0 ms.
`

abdset, 2020-06-29:
Asuuming that you are using Windows 7 or higher:
- Make hiden files to be shown
- Go to users folder
- Go to <Your user name> folder
- Delete all ugs folders in there (.ugs and .ugsplatform).
- Réinstall UGS

breiler, 2020-12-08:
No response from user, closing

closed 2020-12-08 by breiler