winder/Universal-G-Code-Sender #1367: job stops/ugs disconnects at random places(rpi4)

crazyxangel, 2020-03-18:
## Problem description 
UGS disconnects at random times and the job interrupts

### Expected Behavior
for it to complete the program

### Actual Behavior
router randomly stops moving ugs says machine offline

### Steps to Reproduce the Problem
1. open program
2. start a job
3. things go fine i touch nothing the ubs cable doesnt move the pi doesnt move the arduino doesnt move
4. suddenly machine stops moving ugs disconnected
### Specifications
#### Version
UGS Platform 2.0 - Nightly Build Mar 23, 2018

#### Operating system
  Linux on RaspberryPI

#### Platform
 GRBL 1.1f

breiler, 2020-03-18:
Are you running with the JSSC connection driver?
https://github.com/winder/Universal-G-Code-Sender/wiki/Raspberry-Pi#change-the-connection-driver

breiler, 2020-03-18:
I would also recommend upgrading to the nightly build:
https://github.com/winder/Universal-G-Code-Sender#downloads

crazyxangel, 2020-03-18:
i followed that guide for the PI step by step bout a week ago so the connection driver has been changed and it should be the last snapshot (the one you download when you click on "raspberry pi" on the download link)

breiler, 2020-03-19:
The problem with connection driver makes the controller unresponsive but it will still be connected. So after thinking about this it's not likely to be the problem, sorry.

You wrote "UGS Platform 2.0 - Nightly Build Mar 23, 2018" which is pretty old, but it shouldn't behave like this either.

This could be an electrical problem? I've had all kinds of problems when running my spindle causing EMI. Does the same problem occur if you disconnect the power to the steppers and your spindle?

crazyxangel, 2020-03-19:
i'll try running it like that for a couple hours 
but it would be better then to distance the arduino from the drivers and the power supplies of the spindle and drivers?

breiler, 2020-03-19:
If it's possible to move the Arduino to rule out any EMI-problems go for it!

crazyxangel, 2020-03-19:
so my power supplies and stepper drivers are in a powder coated metal box
would it be enough place the arduino on the outside of the box? or would you recommend a bigger distance?

breiler, 2020-03-19:
It completely depends on your setup and in my (rather short) experience, these problems are hard to debug. I've had a complete machine working perfectly up until I turned on the spindle or a water pump, where I'd get ghost signals in my limit switches. To make things worse is that these problems can be intermittent. If it's an electrical problem it could be caused by ground loops or cables not shielded enough.

But before you rip your machine apart, you need to rule out if this is a general problem with connection to Arduino. So if you still have problems running the GCode-program with the Arduino without any peripherals I'd start by replacing the USB-cable (with a good shielding) or try another Arduino unit.

crazyxangel, 2020-03-19:
are those blue shielded usb cables good?

crazyxangel, 2020-03-19:
so with the blue shielded cable on my desktop pc everything runs fine no issues but on the pi its just one problem after another guess i'll just try out a diffirent gcode sender thats more raspberry pi focused

abdset, 2020-03-21:
This issue have nothing to do with UGS or GRBL but could be software side or hardware side issue:

**Software side:**
- Out of date PC drivers such as usb driver and motherboard chipset driver.
- Some antivirus softwares would block UGS from running (Known issue: Like Comodo antivirus).

**Hardware side:**
- Use shielded wires for your motors and connect the shield only to arduino GND (Motor side should not be connected to anything).
- Same for limit switches if you are using them.
- Arduino motors drivers must have a heatsink, without that they get too hot and cut themself from power (Safety feature).
- Do not connect power supply GND with arduino GND since some cheap power supplies causes issues in this case.
- If the power supply and the arduino are in the same metal coated box, put a metal plate in between and connect that to arduino GND (Basicly the arduino should be separated from power supply or spindle power cable when they are in same box).
- Do not "ziptype" arduino wires with any kind of power wires unless they are cable shielded inside the box and that shield should be connected to arduino GND.
- I recommend using DRV8825 motor drivers instead of A4988.
- Power the motors with at lease 4 Amps.
- Always use a Fan to cool down the motors drivers (They get too hot even with heatsink installed).
- Poor USB cable quality can also cause this issue (It happend to somebody before) don'nt trust your cable, get new one.
- Check the arduino usb connector if nothing is broken, i'm talking about the square piece of plastic that holds the 4 usb contacts in the center of the connector (This happend to me, i had to ressolder a new usb plug).

I just finished upgrading my home made machine and i was extremely careful with the shielding and cooling.

After all this said, i mostly suspect that the issue is about usb connection or motors drivers and their heatsinks (Motor drivers must stay cold even if they have a heasink).

abdset, 2020-03-26:
I forget to mention:
Reflashing your arduino bootloader may fix the issue, mostly for cloned ones but rarely for original ones.
This is reported in many forums and cnc/grbl/arduino sites.

Envoy22, 2020-04-27:
Similar problem is ocourring with me. Running grbl on arduino and PC (Windows 10, 4gbRAM, pentium). The UGS program works perfectly until M3 and M5 commands, that freeze all the UGS functions and the program stop working. The program only work again if i close it and open again

iamfallen, 2020-06-28:
I am having a similar issue with the latest platform nightly build on linux. UGS randomly disconnects while running the program. I have tried several gcode files and also tried swapping out the usb cables. The strange thing is that the gcode files work to completion in Candle on the same laptop/machine combination.

breiler, 2020-06-28:
@iamfallen what processor architecture is your computer based on? The problems that happens on RPi is a known issue which happens on processors with ARM-architecture.

iamfallen, 2020-06-28:
@breiler Ah, my laptop is an i5 not ARM.

breiler, 2020-06-28:
@iamfallen you wrote that you have the latest nightly build. What is the version date (shown in the titlebar).

iamfallen, 2020-06-28:
@breiler April 25th 2020

breiler, 2020-12-08:
@iamfallen we got reports that the JSerialComm serial driver could be the problem on all types of Linux machines. Switch to JSSC  instead described here: https://github.com/winder/Universal-G-Code-Sender/wiki/Raspberry-Pi#download-and-start-ugs-platform

@crazyxangel I'm closing this as you wrote it also happens on cncjs.

closed 2020-12-08 by breiler