winder/Universal-G-Code-Sender #1370: Tiny UI elements on high resolution windows laptop.

kthornbloom, 2020-03-22:
## Problem description 
On a high resolution device, (Lenovo Yoga 900 Laptop) some UI elements are too small.

### Expected Behavior
Normal UI sizing

### Actual Behavior
Tiny UI

### Specifications
#### Version
Platform Version 2.0 nightly / Mar 20,2020

#### Operating system
Windows 10

#### Platform
  GRBL 1.1

#### Other

[image]

kthornbloom, 2020-03-22:
FYI, I was able to fix this:

1. Right click ugsplatform.exe
2. Properties
3. Click compatibility tab
4. Use the following settings:
[image]

I'll leave it up to someone else if that's good enough to close this :)

abdset, 2020-03-22:
Good to know how you fixed it, this may help others.
Thanks a lot.

breiler, 2020-12-06:
Closing this issue as there is a workaround. But Netbeans Platform have been updated with a lot of fixes for hdpi-issues that might fix this natively.

closed 2020-12-06 by breiler