winder/Universal-G-Code-Sender #1371: Can't reset window layout on mac Version 2.0 Nightly 3-9-20

lorenrogers, 2020-03-22:
## Problem description 
`Window > Reset Windows` does not work. Two error messages appear.

[image]

### Expected Behavior
When I run reset windows, the app is reset to the default layout I had when the app was first installed.

### Actual Behavior
Error messages are shown (see below) and the windows are unchanged.

### Steps to Reproduce the Problem
1. Open the app
1. "Delete" a window by clicking the x
1. Click `Window > Reset Windows`

### Specifications
#### Version
UGS Platform 2.0 - Nightly Build Mar 9, 2020

#### Operating system

[image]

breiler, 2020-03-22:
Could you attach your log file:
https://github.com/winder/Universal-G-Code-Sender/wiki/Configuration#log-files

lorenrogers, 2020-03-22:
[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/4365570/messages.log)

lorenrogers, 2020-03-22:
Also, it's worth noting that the path is different from what's on the wiki: `~/Library/Application Support/ugsplatform/2.0-SNAPSHOT/dev/var/log`

breiler, 2022-01-14:
Sorry about the delayed response. According to the log the program is missing a bunch of files. Got deleted somehow?
Reinstall and try again.

(Closing as the issue is really old)

closed 2022-01-14 by breiler