winder/Universal-G-Code-Sender #1374: "Stop" function re-sets coordinates after manually setting with G92

gerhick, 2020-03-27:
Hi All,

I'm having a great deal of trouble with UGS at the moment. I had to abort a job because a tool snapped, and caused the stepper motors to skip - thus loosing my current position.
There's 2 problems here. I've figured out that UGS is the cause for one of them - the other I'm suspicious, but not certain.

Firstly the steps to create UGS problem (as per this post title).
- I use G92 to set the X and Y coordinate of the tool by eyeballing features that have already been cut.
- Press "Play" to start the job
- Inevitably something goes wrong with where the job actually starts(second problem I haven't figured out yet)
- Press "Stop" to stop the job, and the X and Y coordinates re-set to some seemingly arbitrary locations which requires me to start the whole process again.

This resetting of the X and Y co-ordinates doesn't occur if I use the "Reset Zero" feature - which I understood was just a G92 command wrapped up in a macro button, so it seems like UGS has some other functions happening when I click the "Stop" button...

Any suggestions on how I can keep the current position while pressing the "Stop" button????? This is killing me at the moment. I literally can't finish my job :'(

Now, second problem is that inevitably something goes wrong after I re-set the current position with G92...
- Set current position with G92
- move to start cutting but notice that the X and Y are offset by some amount - sometimes small, sometimes large.

This problem isn't consistent enough to diagnose properly, so I'm hoping someone recognises the symptoms... 

Under normal conditions the machine works reasonably well. If I'm lucky enough for a job to run start to finish without the need to stop it, then it works fine. Only if I have to re-set the home position do I have these problems. 

I'm hoping someone can assist. I'm sick of scrapping material...

Thanks

Gerrard

breiler, 2020-03-27:
@winder you may correct me if I'm wrong. =)

When we stop the job we also reset GRBL to clear any queued commands in the controller. Otherwise the stop wouldn't be immediate. The reset will also clear the G92 position, see #501.

I'm wondering why you are using G92 though. What version of GRBL are using?
In version 9 or later another command is used for resetting coordinates to zero.

It's also interesting what version of UGS you are running.

gerhick, 2020-03-27:
Thanks for the quick response... 

I suspected that was the case RE: the stop function.

I'm using G92 because it does what I want, and I didn't know there was an alternative - what is the alternative? Will it help to resolve this issue?

I'm not actually sure what version of GRBL i'm using - it's an old 2nd hand machine that I purchased. I know GRBL is out of date though, I just don't know how far out of date.

UGS is the latest version of the platform. I updated that in the last few days.

Regards

breiler, 2020-03-27:
You can run the command $I which will output something like:
```
>>> $I
[VER:1.1h.20190825:]
[OPT:V,15,128]
ok
```

The alternative is to use G54-G59 instead. Check out the issue #501 which links some documentation. Also the documentation over at GRBL: https://github.com/gnea/grbl/wiki/Grbl-v1.1-Commands

abdset, 2020-03-28:
_I hope i'm not out of topic here (My english is not perfect)_

> This resetting of the X and Y co-ordinates doesn't occur if I use the "Reset Zero" feature - which I understood was just a G92 command wrapped up in a macro button, so it seems like UGS has some other functions happening when I click the "Stop" button...

You can read on console what UGS is sending to grbl when you click on RESET ZERO, something like:
>>> G10 P0 L20 X0 Y0 Z0

I have some macros assigned to my CNC controller buttons and i'm using the same command to RESET ZERO and worked perfectly to me.

If you click on Return to zero, you'll see those commands in console:
>>> G21G90 G0Z5
>>> G90 G0 X0 Y0
>>> G90 G0 Z0

_(G21 stands for millimiters unit, if you are using inches put G20 instead)._

If your issue is resolved, please give us a feedback.

closed 2020-04-13 by breiler