winder/Universal-G-Code-Sender #1377: Probe Module G90/G91

Cybnew, 2020-03-31:
Running the probe module, the last G-Code command issued is G91. This leaves the machine in relative coordinates which for me, is not beneficial to a machining workflow. Is this by design? Setting back to g90 works perfectly for me. Is it possible to have a setting which allows you to choose the default?

Thanks

JacksCrafts, 2020-04-01:
Couldn't you just make a Macro button to say G90.

Cybnew, 2020-04-03:
> Couldn't you just make a Macro button to say G90.

Yes, and that’s what I’ve done, but it seems like a poor use case for a probe module if you’re going to probe and then use relative coordinates based off the position of the tool head. The question came up again by someone else in the UGS google group today.

Hanier, 2022-06-14:
New to CNC with my first machine EVER.  I'm doing a lot of fiddling, but it's all fun.  I just discovered the probe module leaves the system in G91, and it almost cost me a collision, so I thought about what might improve the situation.  My initial reaction was that the status ought to get taken in at the beginning of the probe sequence, and those states should be returned leaving.  That way the machine is left in the same manual state as before.

P.S.  Thanks to all you who do this software wizardry.

breiler, 2023-08-16:
Closing, the gcode state is now restored after probing is done.

closed 2023-08-16 by breiler