winder/Universal-G-Code-Sender #1383: Error while processing response <<Idle|MPos:Srº‚‚b’Šrººšbºr‚ººâ	™é15,128|FS:0,0>>

de3tri, 2020-04-04:
## Problem description 
Hello, I have a CNC 3018, and I keep getting "error while processing response". 
On laptop i run ugs platform and I don't get the error it just stops moving, I have to disconnect the usb to regain control. 
On pc i run ugs 64 and I get the error and sometimes it stops, but not so often. 

I have tried Candle and LiteFire for laser, same problem, it just stops. 

I grounded my laptop on usb, I grounded cnc frame and spindle supply. Same problem.. 
I reinstalled usb driver CH3xx. 

I have read all the forums I found, still no solution. 

What can I do to fix this, any suggestions or ideas are more then welcome. 

My config is: 

#### Operating system
UGS Platform v2 nightly build apr 23 2019
windows 10 86 on laptop
windows 10 64 on pc

#### Platform
[VER:1.1h.20190825:]
[OPT:V,15,128]

>>> $#
[G54:-0.700,21.705,-7.923]
[G55:0.000,0.000,0.000]
[G56:0.000,0.000,0.000]
[G57:0.000,0.000,0.000]
[G58:0.000,0.000,0.000]
[G59:0.000,0.000,0.000]
[G28:0.000,0.000,0.000]
[G30:0.000,0.000,0.000]
[G92:0.000,0.000,0.000]
[TLO:0.000]
[PRB:0.000,0.000,0.000:0]

>>> $g
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F270 S10000]

$0 = 10    (Step pulse time, microseconds)
$1 = 25    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 5    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 0    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 3    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 0    (Soft limits enable, boolean)
$21 = 0    (Hard limits enable, boolean)
$22 = 0    (Homing cycle enable, boolean)
$23 = 3    (Homing direction invert, mask)
$24 = 25.000    (Homing locate feed rate, mm/min)
$25 = 500.000    (Homing search seek rate, mm/min)
$26 = 250    (Homing switch debounce delay, milliseconds)
$27 = 1.000    (Homing switch pull-off distance, millimeters)
$30 = 1000    (Maximum spindle speed, RPM)
$31 = 0    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$100 = 800.000    (X-axis travel resolution, step/mm)
$101 = 800.000    (Y-axis travel resolution, step/mm)
$102 = 401.314    (Z-axis travel resolution, step/mm)
$110 = 500.000    (X-axis maximum rate, mm/min)
$111 = 500.000    (Y-axis maximum rate, mm/min)
$112 = 500.000    (Z-axis maximum rate, mm/min)
$120 = 250.000    (X-axis acceleration, mm/sec^2)
$121 = 250.000    (Y-axis acceleration, mm/sec^2)
$122 = 50.000    (Z-axis acceleration, mm/sec^2)
$130 = 285.000    (X-axis maximum travel, millimeters)
$131 = 193.000    (Y-axis maximum travel, millimeters)
$132 = 150.000    (Z-axis maximum travel, millimeters)

#### Error message

<Idle|MPos:79.700,21.773,7.077|Bf:15,128|FS:0	òj
Error while processing response <<Idle|MPos:Srº‚‚b’Šrººšbºr‚ººâ	™é15,128|FS:0,0>>
<Idle|MPos:79.700,21.773,7.077|Bf:15,L&âMé0,â]
=é-0é‚‚b’Šrº‚ªbjºrÊ’šòj
<Idle|MPos:79.700,21.773,7.077|Bf:15,128|FS:0,0j
Error while processing response <<Idle|MPos:Srº‚‚b’Šrººšbºr‚ººâ	™é15,128|FS:0,0>>
Error while processing response <<Idle|MPos:79.700,21éºšbºr‚ººâ	™é15,128|FS:0,0>>
<Idle|MPos:79.700,21.773,7.077|Bf:15,128|FS:b‚òj
<Idle|MPos:79.700,21.773,7.077|Bf:15,128|Ô¥‚b‚òj
Error while processing response <<Idle|MPos:79.70b’Šrººšbºr‚ººâ	™é15,128|FS:0,0|Ov:200,100,100>>
Error while processing response <<Idle|MPos:79.700,21.ÓÓbºr‚ººâ	™é15,128|FS:0,0>>
Error while processing response <<Idle|MPos:7Né‚‚b’Šrººšbºr‚ººâ	™é15,128|FS:0,0|Ov:200,100,100>>
Error while processing response <<Idle|MPos:79.700,21éºšbºr‚ººâ	™é15,128|FS:0,0>>

breiler, 2020-04-04:
It sounds like you also suspect electromagnetic interference as you mentioned the grounding. Just make sure you have not created any ground loops which would act as antennas.

I would start by checking the usb cable and make sure it is thick and shielded.

Does this occur if you do not run the spindle or stepper motor drivers?

de3tri, 2020-04-04:
I have only 2 cables, it's 1-1.2 m long.. both of them. 
At this moment it's connected to PC and it's not grounded in any way, but how is it a ground loop that acts like a antenna? 
Yes it is happening also while stationary, as you see I have posted Idle mpos..

khturnings, 2020-04-04:
I have had a similar issue, I believe I have resolved it in 2 ways -
1. I replaced my cheap Arduino which had a CH3XX usb driver chip with one that has ATMEGA16U2 chip - https://www.ebay.co.uk/itm/173653258248
This was a partial fix i.e. improved performance but still had some dropouts and freezes
2. I replaced my cheap  2m usb cable with a good quality 3m usb cable which also has a built in signal repeater and this seems to have fixed my issues. But I have yet to test on a long running job.

Maybe this will help you de3tri

de3tri, 2020-04-04:
I'v read about chip 16u2, if has the last firmware it wont make any more problems, I was looking where to buy one in my country.. 
What is so strange to my is that weird characters in the error..

abdset, 2020-04-06:
Are you using a CNC pendant connected somewhere on the controller ?
The manual say dont connect USB to the PC and use pendant at the same time.
I dont have that machine but i gess they are sharing the same port.
Also you may:
- Check the baudrate at USG platform connection section (top left on the screen) set it to 115200.
- Have you choosed the right controller firmware on UGS Platform ? (GRBL or  smouthieware or tinyG or so ..) according to your machine. Most people are using GRBL.
- Try to find the micro-controller bootloader file on the net and reflash it, some people get their connection issues fixed by doing it.

abdset, 2020-04-06:
DRV8825 has a feature that will cut it from power when its too hot before it gets damaged.
This mean a quick job will be fine without a fan but longer job will take time and need a fan to cool them down. You also need to put heatsinks.
They are a very good drivers, i'm stressing them a lot and never have issues.
Please let us know what did fix your issue when done. We would like to help others if needed.
Until you get your new parts, you could try to flash the bootloader and see how it goes, if its a succes you will have two working controllers.

de3tri, 2020-04-08:
So my new Arduino uno with drv8825 had arrived and I installed them, they work like a charm, no errors, no problems so far, and if until now my max speed was 1200 mm/min (at 1500mm would freeze) now I work with it at 1500-1800mm/min, (at 2500 it freezes). 

Now problem nr2, to connect the laser to the shield, can I use a different power supply and use only one cable TTL or do I need a step down dc-dc so the GND is common ?
The "manual/internet" says that Z+ is now tll, and If I measure Z+ and GND it is, if I set laser power to 100% I have 5V, as expected.. but so does Z-, Y,X, SPNEN,SPDDR etc and all of the pins ... and that is werid.

abdset, 2020-04-08:
- For the feedrate, if you have more amps and volts for your motors you can then reach higher speed. I am using a laptop power supply whitch delivers 19 volts at 4.7 amps and my machine can reach 10000 mm/min without any issue. I use 7808 to pull down the voltage to 8 volts whitch i am using for the arduino uno (connected to its power plug), and 19 volts goes directly to the cnc shield for the motors.
- I dont have experience with laser though, so i think you will need somebody else to help you with that.
Wish you the best.

de3tri, 2020-04-08:
My power supply is 24v, 4Amps, but as I was saying, with arduino nano and A4988 drivers my max speed was 1200mm/min, now it's much better. Why do you power aruduino uno? It's not connected via usb?

khturnings, 2020-04-08:
@de3tri good you god things working smoothly. You can power via USB or independently.

For my machine I use 36v to power machine and an old ATX power supply to provide 12v for my laser.
I also use the 5v rail to power some low powered laser crosshairs for aligning my router.
I will look later at my Laser -> Arduino connections and post what they are.

abdset, 2020-04-08:
No i'm not using usb on arduino, i have added ESP32 connected to arduino via RX/TX and have done a program that will allow me use my machine via bluetooth with joystick jogging feature.
That's why i need to power the arduino (no usb cable connected to it).

breiler, 2020-04-15:
Closing issue as it was resolved using a new arduino.

closed 2020-04-15 by breiler