winder/Universal-G-Code-Sender #1386: Crash on Mac with security violation error - Java 14

MikeMitterer, 2020-04-09:
Im using Mojave 10.14.6 with several versions of Java on my machine.

Lately (in the meantime I've installed Java 14) I got a crash report and a Java security error.
The error was something like "Not allowed to  run application without certificate..."

I think ugs should be aware of JAVA_HOME | JDK_HOME...

However - I was able to solve the problem by:

- in my .bashrc:
    export JAVA8_HOME=`/usr/libexec/java_home -v 1.8`
    export JAVA_HOME=$JAVA8_HOME
    export JDK_HOME=$JAVA_HOME

-  adding the following line to `ugsplatform`:

``` bash
...
PRG=$0

#### I added these line ###
if [ -z "${jdkhome}" ]; then
    jdkhome=${JDK_HOME}
    if [ -z "${jdkhome}" ]; then
      jdkhome=$JAVA_HOME
    fi
fi

while [ -h "$PRG" ]; do
    ls=`ls -ld "$PRG"`
    link=`expr "$ls" : '^.*-> \(.*\)$' 2>/dev/null`
    if expr "$link" : '^/' 2> /dev/null >/dev/null; then
	PRG="$link"
    else
	PRG="`dirname "$PRG"`/$link"
    fi
...
```

### Specifications
#### Version
I tried both UGS nightly (2020 04 06) and Stable 2019 08 14

#### Operating system
Mac Mojave 10.14.6

winder, 2020-04-09:
There are some comments about this in #1328 

Thats interesting that overriding the jdkhome is part of the problem. Instead of setting `jdkhome` in the launcher script (which is a build artifact) would suggest setting it in `ugsplatform.conf`.

There is also a UGS package which ships with the java runtime (and uses jdkhome), which might work without any changes: https://github.com/winder/Universal-G-Code-Sender#downloads

jeffpoulton, 2020-06-07:
I tried running UGS on OS X Catalina with both Java 11 and Java 14. Both versions of Java resulted in the UGS crashing. I installed Java 8 and followed @MikeMitterer's steps and was able to successfully launch UGS without it crashing.

However, I did not want to set Java on my machine to 1.8. Following advice from @winder to configure `jdkhome` in `ugsplatform.conf`, this line does the trick for me:

`jdkhome="`/usr/libexec/java_home -v 1.8`"`

My machine is configured to use Java 14, but UGS is running against Java 8.

erichVK5, 2020-07-13:
I was successfully running UGS on this netbook with ubuntu 16LTS, but after upgrading to 20.04 I am getting similar errors to others:

OpenJDK 64-Bit Server VM warning: Options -Xverify:none and -noverify were deprecated in JDK 13 and will likely be removed in a future release.
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by org.netbeans.ProxyURLStreamHandlerFactory (file:/home/tonia/Source/ugsplatform-linux/platform/lib/boot.jar) to field java.net.URL.handler
WARNING: Please consider reporting this to the maintainers of org.netbeans.ProxyURLStreamHandlerFactory
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release

This is happening with the 2.0 platform snapshot, as well as with the August 14, 2019 stable release.

The August 14 release did launch, but it exits with the above error on trying to launch the visualiser screen. 

The 2.0 platform release hangs while displaying the launch screen and "loading modules" while trying to start up.

I have tried modifying the ugsplatform.conf file in the 2.0 platform snapshot as described above, but this did not work despite trying various combinations for the following local default java directories:

$ ls /usr/lib/jvm/
default-java  java-1.11.0-openjdk-amd64  java-11-openjdk-amd64  openjdk-11
 
Any advice would be appreciated.

Erich.

closed 2020-07-17 by breiler