winder/Universal-G-Code-Sender #1391: UGSplatform wont start under Ubuntu 20.04

ouinouin, 2020-04-16:
## Problem description 
While launching UGS platform under ubtuntu 20.04, i get the following : 
[image]
the screen locks at this step, and goes no further.
it looks like the same issue  than : https://github.com/winder/Universal-G-Code-Sender/issues/1228
but since this bug report was closed as working ...
### Expected Behavior
The UGS platform shall not lock at startup.

### Actual Behavior
UGS doesnt finish the startup and hangs at splash screen

### Steps to Reproduce the Problem
 1. Open the program

### Specifications
#### Version

#### Operating system

#### Platform

breiler, 2020-04-16:
Which version of UGS are you running?

Did you install Java yourself or are you using the version that comes bundled with Java from here?
https://github.com/winder/Universal-G-Code-Sender#downloads

ouinouin, 2020-04-17:
Hi, yes i took the bundled version and tried both the latest version and the stable version... I put the version numbers in my first message but apparently was eaten by github ....
so the error is the same with https://ugs.jfrog.io/ugs/UGS/nightly/ugs-platform-app-2.0-SNAPSHOT-linux.tar.gz
and with : http://bit.ly/2Z3DIVC (the 2.0 platform beta in the stable list of downloads.)

breiler, 2020-04-18:
Can you upload your log file:
https://github.com/winder/Universal-G-Code-Sender/wiki/Configuration#log-files

ouinouin, 2020-04-19:
hi, here is the latest log

`>Log Session: Sunday, April 19, 2020 at 8:31:23 PM Central European Summer Time
>System Info: 
  Product Version         = Universal Gcode Platform 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54
  Operating System        = Linux version 5.4.0-21-generic running on amd64
  Java; VM; Vendor        = 13.0.1; OpenJDK 64-Bit Server VM 13.0.1+9; AdoptOpenJDK
  Runtime                 = OpenJDK Runtime Environment 13.0.1+9
  Java Home               = /home/ouinouin/ugsplatform-linux/jdk/jdk-13.0.1+9-jre
  System Locale; Encoding = fr_FR (ugsplatform); UTF-8
  Home Directory          = /home/ouinouin
  Current Directory       = /home/ouinouin/ugsplatform-linux/bin
  User Directory          = /home/ouinouin/.ugsplatform/2.0-SNAPSHOT/dev
  Cache Directory         = /home/ouinouin/.ugsplatform/2.0-SNAPSHOT/dev/var/cache
  Installation            = /home/ouinouin/ugsplatform-linux/etc
                            /home/ouinouin/ugsplatform-linux/ide
                            /home/ouinouin/ugsplatform-linux/bin
                            /home/ouinouin/ugsplatform-linux/ugsplatform
                            /home/ouinouin/ugsplatform-linux/extra
                            /home/ouinouin/ugsplatform-linux/java
                            /home/ouinouin/ugsplatform-linux/platform
                            /home/ouinouin/ugsplatform-linux/platform
  Boot & Ext. Classpath   = 
  Application Classpath   = /home/ouinouin/ugsplatform-linux/platform/lib/boot.jar:/home/ouinouin/ugsplatform-linux/platform/lib/org-openide-modules.jar:/home/ouinouin/ugsplatform-linux/platform/lib/org-openide-util.jar:/home/ouinouin/ugsplatform-linux/platform/lib/org-openide-util-lookup.jar:/home/ouinouin/ugsplatform-linux/platform/lib/org-openide-util-ui.jar
  Startup Classpath       = /home/ouinouin/ugsplatform-linux/platform/core/org-openide-filesystems-compat8.jar:/home/ouinouin/ugsplatform-linux/platform/core/asm-all-5.0.1.jar:/home/ouinouin/ugsplatform-linux/platform/core/org-netbeans-libs-asm.jar:/home/ouinouin/ugsplatform-linux/platform/core/org-openide-filesystems.jar:/home/ouinouin/ugsplatform-linux/platform/core/core-base.jar:/home/ouinouin/ugsplatform-linux/platform/core/core.jar:/home/ouinouin/ugsplatform-linux/ugsplatform/core/locale/core_ugsplatform.jar
-------------------------------------------------------------------------------
INFO [com.willwinder.ugs.nbp.core.lifecycle.startup]: Loading LocalizingService...
INFO [com.willwinder.universalgcodesender.utils.SettingsFactory]: Log location: /home/ouinouin/.ugs/UniversalGcodeSender.json
INFO [com.willwinder.universalgcodesender.utils.SettingsFactory]: Loading settings.
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Applying settings.
INFO [com.willwinder.ugs.nbp.core.lifecycle.startup]: Loading JogService...
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Adding controller listener: ContinuousJogWorker
WARNING [org.netbeans.modules.autoupdate.ui.actions.AutoupdateSettings]: The property "netbeans.default_userdir_root" was not set!
INFO [org.netbeans.core.startup.NbEvents]: Turning on modules:
	org.openide.util.lookup [8.39 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54]
	org.openide.util [9.13 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54]
	org.openide.util.ui [9.14 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54]
	org.openide.modules [7.54 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54]
	org.netbeans.api.annotations.common/1 [1.33 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54]
	org.openide.filesystems [9.16 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54]
	org.openide.awt [7.74 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54]
	org.netbeans.api.progress/1 [1.53 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54]
	org.netbeans.api.progress.nb [1.53 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54]
	org.openide.dialogs [7.48 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54]
	org.openide.nodes [7.51 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54]
	org.openide.windows [6.83 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54]
	org.netbeans.modules.editor.mimelookup/1 [1.46 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54]
	org.openide.text [6.73 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54]
	org.netbeans.api.scripting [1.3 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54]
	org.netbeans.swing.tabcontrol [1.62 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54]
	org.netbeans.swing.outline [1.40 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54]
	org.openide.explorer [6.68 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54]
	org.openide.actions [6.45 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54]
	org.netbeans.modules.queries/1 [1.49 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54]
	org.netbeans.api.templates [1.14 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54]
	org.openide.filesystems.nb [9.16 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54]
	org.openide.loaders [7.73 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54]
	org.netbeans.api.intent [1.8 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54]
	org.netbeans.api.io [1.9 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54]
	org.openide.io [1.55 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54]
	org.openide.filesystems.compat8 [9.15 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54]
	org.netbeans.swing.plaf [1.47 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54]
	org.netbeans.spi.tasklist/1 [1.41.0.1 1 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54]
	org.netbeans.spi.quicksearch [1.33 11.2-d04fb24027334c4b6fd8397b5d0cdd33187a8f54]
[... cut ...]
INFO [com.willwinder.universalgcodesender.utils.FirmwareUtils]: /resources/firmware_config/test_2.json
INFO [com.willwinder.universalgcodesender.utils.FirmwareUtils]: /resources/firmware_config/tinyg.json
INFO [com.willwinder.universalgcodesender.utils.FirmwareUtils]: /resources/firmware_config/grbl.json
INFO [com.willwinder.universalgcodesender.utils.FirmwareUtils]: /resources/firmware_config/smoothieboard.json`

i tried to clear the .usgplatform in my home folder with no success.
ill try with a fresh ubuntu 20.04 install in a VM if I have time

winder, 2020-04-25:
Try removing the `.ugs` directory as well

tbfunk, 2020-05-11:
I havent been able to get UGS to run without using sudo.
"./ugsplatform"   always hangs at modules
"sudo ./ugsplatform" works but I have random connection/disconnection errors occasionally

bianchifan, 2020-05-16:
While UGSPlatform's nightly from Febr.27 was runnin' fine in Ubuntu 19.10 it shows only a weired screen on 20.04

[image]
The actual nitely stucks on splash as shown above..:((
Here is the latest log:
[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/4638275/messages.log)

ngocminhdao88, 2020-06-01:
I am using a nighty build with bundled java on Ubuntu 20.04. By removing .ugs and .ugsplatform at ~, i can run the program.

erichVK5, 2020-07-13:
I was successfully running a maven built UGS on this netbook with ubuntu 16LTS, but after upgrading to 20.04 I am getting similar errors to others:

OpenJDK 64-Bit Server VM warning: Options -Xverify:none and -noverify were deprecated in JDK 13 and will likely be removed in a future release.
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by org.netbeans.ProxyURLStreamHandlerFactory (file:/home/tonia/Source/ugsplatform-linux/platform/lib/boot.jar) to field java.net.URL.handler
WARNING: Please consider reporting this to the maintainers of org.netbeans.ProxyURLStreamHandlerFactory
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release

This is happening with the 2.0 platform snapshot, as well as with the August 14, 2019 stable release.

The August 14 release did launch, but it exits with the above error on trying to launch the visualiser screen.

The 2.0 platform release hangs while displaying the launch screen and "loading modules" while trying to start up.

I have tried modifying the ugsplatform.conf file in the 2.0 platform snapshot as described in the mac os security error issue, but this did not work despite trying various combinations for the following local default java directories:

$ ls /usr/lib/jvm/
default-java java-1.11.0-openjdk-amd64 java-11-openjdk-amd64 openjdk-11

Similar issues appear to be afflicting java apps built with certain Java3d versions, i.e. see

http://forum.jogamp.org/JAVA-11-quot-illegal-reflective-access-quot-warning-td4039322.html
https://forum.image.sc/t/java3d-1-6-0-and-the-near-future/19471/2

and some more gory details here:

https://www.docjava.com/pub/document/reports/joglAndJava3d.pdf

Any advice would be appreciated.

Erich.

larsskj, 2021-02-10:
I'm hit by the same trying to make UGS run on Ubuntu 20.04.2 LTS. I've tested the latest snapshot in both versions - with and without embedded Java. I won't run at all using my own Java - OpenJDK 11.0.10, but it does run using the embedded Java. I tested 2.0.7-SNAPSHOT.

My last attempt looked like this:

[image]

It ran for a while, then crashed with this message:

```
OpenJDK 64-Bit Server VM warning: Options -Xverify:none and -noverify were deprecated in JDK 13 and will likely be removed in a future release.
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by org.netbeans.ProxyURLStreamHandlerFactory (file:/opt/ugsplatform-linux-2.0.7-SNAPSHOT/platform/lib/boot.jar) to field java.net.URL.handler
WARNING: Please consider reporting this to the maintainers of org.netbeans.ProxyURLStreamHandlerFactory
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0x00007f81af4b152d, pid=125242, tid=125306
#
# JRE version: OpenJDK Runtime Environment (13.0.1+9) (build 13.0.1+9)
# Java VM: OpenJDK 64-Bit Server VM (13.0.1+9, mixed mode, sharing, tiered, compressed oops, g1 gc, linux-amd64)
# Problematic frame:
# C  [iris_dri.so+0xc0d52d]
#
# Core dump will be written. Default location: Core dumps may be processed with "/usr/share/apport/apport %p %s %c %d %P %E" (or dumping to /home/lars/core.125242)
#
# An error report file with more information is saved as:
# /home/lars/hs_err_pid125242.log
#
# If you would like to submit a bug report, please visit:
#   https://github.com/AdoptOpenJDK/openjdk-build/issues
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#
/opt/ugsplatform-linux/bin/../platform/lib/nbexec: line 425: 125242 Aborted                 (core dumped) "/opt/ugsplatform-linux/jdk/jdk-13.0.1+9-jre/bin/java" -Djdk.home="/opt/ugsplatform-linux/jdk/jdk-13.0.1+9-jre" -classpath "/opt/ugsplatform-linux/platform/lib/boot.jar:/opt/ugsplatform-linux/platform/lib/org-openide-modules.jar:/opt/ugsplatform-linux/platform/lib/org-openide-util.jar:/opt/ugsplatform-linux/platform/lib/org-openide-util-lookup.jar:/opt/ugsplatform-linux/platform/lib/org-openide-util-ui.jar" -Dnetbeans.running.environment=gnome -Dnetbeans.dirs="/opt/ugsplatform-linux/etc:/opt/ugsplatform-linux/ide:/opt/ugsplatform-linux/bin:/opt/ugsplatform-linux/ugsplatform:/opt/ugsplatform-linux/extra:/opt/ugsplatform-linux/java:/opt/ugsplatform-linux/platform:" -Dnetbeans.home="/opt/ugsplatform-linux/platform" '-Xms64m' '-Xverify:none' '-Dsun.java2d.noddraw=true' '-Dsun.awt.noerasebackground=true' '-Dnetbeans.indexing.noFileRefresh=true' -XX:+HeapDumpOnOutOfMemoryError -XX:HeapDumpPath="/home/lars/.ugsplatform/2.0.7-SNAPSHOT/dev/var/log/heapdump.hprof" org.netbeans.Main --cachedir "/home/lars/.ugsplatform/2.0.7-SNAPSHOT/dev/var/cache" --userdir "/home/lars/.ugsplatform/2.0.7-SNAPSHOT/dev" "--branding" "ugsplatform" 0<&0
```
Deleting `~/.ugs` and `~/.ugsplatform` solves the problem for a while - but also deletes all my settings.

breiler, 2022-12-15:
Please try the latest nightly build, I have upgraded to the latest OpenGL library (or version 2.0.13 when it get released)

closed 2022-12-15 by breiler