winder/Universal-G-Code-Sender #1392: UGS Platform won't load on Lubuntu 18.04

UWBernie, 2020-04-16:
I'm attempting to run the UGS Platform on an old netbook running Lubuntu 18.04 The splash screen loads up and for a moment I see the window then it closes.  This was using the last stable build, and java (openjdk version "1.8.0_242"). Note I also tried using the linked oracle java 1.8.0_251.  Despite the warnings, the program doesnt terminate until the bolded line

bernie@Bernie-1000HE:/Downloads/ugsplatform/bin$ ./ugsplatform
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by org.netbeans.ProxyURLStreamHandlerFactory (file:/home/bernie/Downloads/ugsplatform/platform/lib/boot.jar) to field java.net.URL.handler
WARNING: Please consider reporting this to the maintainers of org.netbeans.ProxyURLStreamHandlerFactory
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
**Inconsistency detected by ld.so: dl-lookup.c: 111: check_match: Assertion `version->filename == NULL || ! _dl_name_match_p (version->filename, map)' failed!**
bernie@Bernie-1000HE:/Downloads/ugsplatform/bin$ java -version
openjdk version "1.8.0_242"
OpenJDK Runtime Environment (build 1.8.0_242-8u242-b08-0ubuntu3~18.04-b08)
OpenJDK Server VM (build 25.242-b08, mixed mode)

Using the nightly build Linux version results in the following error, I'm guessing this is because I'm using a 32 bit os and processor and its expecting 64 bit?

bernie@Bernie-1000HE:/Downloads/ugsplatform-linux/bin$ ./ugsplatform
./../platform/lib/nbexec: line 421: /home/bernie/Downloads/ugsplatform-linux/jdk/jdk-13.0.1+9-jre/bin/java: cannot execute binary file: Exec format error

winder, 2020-04-25:
This seems to be an issue with the Java runtime. I was able to reproduce this issue on ubuntu 18.04 using `openjdk version "11.0.6" 2020-01-14`

The version of UGS with a bundled JRE worked fine, please give the version here a try: https://github.com/winder/Universal-G-Code-Sender#downloads

breiler, 2020-06-23:
Closing as we didn't get response from user

closed 2020-06-23 by breiler