winder/Universal-G-Code-Sender #1393: Raspberry Pi - UGS-Platform-App Failure

johan42323, 2020-04-17:
Hi,

I am running a new, clean version of Raspbian on my Raspberry pi3 built from NOOBS.

I followed the steps as mentioned in:
https://github.com/winder/Universal-G-Code-Sender/wiki/Raspberry-Pi
(Under: Building on Raspberry Pi)

The only difference was the java -version (shows 11.0.5, on mine 11.0.6)

...............

I am getting the following failure:

[INFO]
[INFO]
[ERROR] Tests run: 1, Failures: 1, Errors: 0, Skipped: 0
[INFO]
[INFO] ------------------------------------------------------------------------
[INFO] Reactor Summary for ugs-parent 2.0-SNAPSHOT:
[INFO]
[INFO] ugs-parent .........................................                        SUCCESS [  3.104 s]
[INFO] ugs-core ...........................................                          SUCCESS [01:48 min]
[INFO] ugs-pendant ........................................                      SUCCESS [  7.239 s]
[INFO] ugs-cli ............................................                             SUCCESS [  1.482 s]
[INFO] ugs-classic ........................................                         SUCCESS [ 19.947 s]
[INFO] ugs-platform-parent ................................                SUCCESS [  1.035 s]
[INFO] ugs-platform-branding ..............................             SUCCESS [ 10.794 s]
[INFO] ugs-platform-ugslib ................................                SUCCESS [ 44.064 s]
[INFO] ugs-platform-ugscore ...............................              SUCCESS [ 19.325 s]
[INFO] ugs-platform-visualizer ............................               SUCCESS [ 16.910 s]
[INFO] ugs-platform-plugin-workflow .......................       SUCCESS [  7.670 s]
[INFO] ugs-platform-gcode-editor ..........................          SUCCESS [ 19.361 s]
[INFO] ugs-platform-surfacescanner ........................         SUCCESS [ 11.522 s]
[INFO] ProbeModule ........................................                     SUCCESS [ 16.437 s]
[INFO] DowelModule ........................................                     SUCCESS [01:02 min]
[INFO] GcodeTools .........................................                        SUCCESS [  9.168 s]
[INFO] ugs-platform-plugin-jog ............................              SUCCESS [ 33.206 s]
[INFO] ugs-platform-plugin-setup-wizard ...................     SUCCESS [ 12.306 s]
[INFO] ugs-platform-welcome-page ..........................        SUCCESS [  8.789 s]
[INFO] ugs-platform-plugin-joystick .......................           SUCCESS [ 51.103 s]
[INFO] ugs-platform-plugin-cloud-storage ..................    SUCCESS [ 49.788 s]
[INFO] ugs-platform-app ...................................                   FAILURE [01:45 min]
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE

//---------------------
Additional Info:
//---------------------

[INFO]
[INFO] Results:
[INFO]
[ERROR] Failures:
[ERROR] com.willwinder.ApplicationTest.testApplication
[INFO]   Run 1: PASS
[ERROR]   Run 2: ApplicationTest.testApplication NbModuleSuite has been started with failOnMessage(WARNING) and failOnException(INFO). The following failures have been captured:
[org.openide.util.RequestProcessor] THREAD: On Start/Stop MSG: Error in RequestProcessor com.willwinder.ugs.nbp.joystick.Startup
java.lang.UnsatisfiedLinkError: /tmp/jnigen/923140537/libjamepadArm.so: libbcm_host.so: cannot open shared object file: No such file or directory
  java.base/java.lang.ClassLoader$NativeLibrary.load0(Native Method)
  java.base/java.lang.ClassLoader$NativeLibrary.load(ClassLoader.java:2430)
  java.base/java.lang.ClassLoader$NativeLibrary.loadLibrary(ClassLoader.java:2487)
  java.base/java.lang.ClassLoader.loadLibrary0(ClassLoader.java:2684)
  java.base/java.lang.ClassLoader.loadLibrary(ClassLoader.java:2617)
  java.base/java.lang.Runtime.load0(Runtime.java:767)
  java.base/java.lang.System.load(System.java:1834)
  com.badlogic.gdx.jnigen.JniGenSharedLibraryLoader.loadLibrary(JniGenSharedLibraryLoader.java:105)
  com.badlogic.gdx.jnigen.JniGenSharedLibraryLoader.load(JniGenSharedLibraryLoader.java:191)
  com.studiohartman.jamepad.ControllerManager.<init>(ControllerManager.java:65)
  com.studiohartman.jamepad.ControllerManager.<init>(ControllerManager.java:42)
[... 85 lines cut ...]
[INFO] ugs-platform-plugin-joystick ....................... SUCCESS [ 51.103 s]
[INFO] ugs-platform-plugin-cloud-storage .................. SUCCESS [ 49.788 s]
[INFO] ugs-platform-app ................................... FAILURE [01:45 min]
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time:  10:30 min
[INFO] Finished at: 2020-04-17T19:13:17+02:00
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal org.apache.maven.plugins:maven-surefire-plugin:3.0.0-M4:test (default-test) on project ugs-platform-app: There are test failures.
[ERROR]
[ERROR] Please refer to /home/pi/Universal-G-Code-Sender-master/ugs-platform/application/target/surefire-reports for the individual test results.
[ERROR] Please refer to dump files (if any exist) [date].dump, [date]-jvmRun[N].dump and [date].dumpstream.
[ERROR] -> [Help 1]
[ERROR]
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR]
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoFailureException
[ERROR]
[ERROR] After correcting the problems, you can resume the build with the command
[ERROR]   mvn <goals> -rf :ugs-platform-app

Thank you in advance!

breiler, 2020-04-17:
Are you trying to build this from a SSH-session?

Try inactivating the tests:
https://stackoverflow.com/questions/1607315/build-maven-project-without-running-unit-tests

johan42323, 2020-04-18:
Hi,

Thank you for your response.
Yes, I am using putty to ssh to the currently headless running Pi. (later it will get a touch screen)

I had a look at your link and ran the following:

mvn -Dmaven.test.skip=true install -P '!create-pendant-web-ui'

I did now get a Build success :)

Again, thank you.

closed 2020-04-18 by breiler