winder/Universal-G-Code-Sender #1395: Freezing in Visualizer

remi31000, 2020-04-22:
## Problem description 
UGS Platform hangup on visualizer tab

### Actual Behavior

Freeze and log showing :

Caused by: com.jogamp.opengl.GLException: Profile GL4bc is not available on X11GraphicsDevice[type .x11, connection :0, unitID 0, handle 0x7f1794494680, owner true, ResourceToolkitLock[obj 0x1d490b7b, isOwner true, <7e837f08, 4199544a>[count 1, qsz 0, owner <AWT-EventQueue-0>]]], but: [GLProfile[GLES1/GLES1.hw], GLProfile[GLES2/GLES3.hw], GLProfile[GL2ES1/GLES1.hw], GLProfile[GL4ES3/GL4.hw], GLProfile[GL2ES2/GL4.hw], GLProfile[GL4/GL4.hw], GLProfile[GLES3/GLES3.hw], GLProfile[GL4/GL4.hw], GLProfile[GL3/GL4.hw], GLProfile[GL2GL3/GL4.hw]]
	at com.jogamp.opengl.GLProfile.get(GLProfile.java:991)
	at jogamp.opengl.GLContextImpl.verifyInstance(GLContextImpl.java:1471)
	at jogamp.opengl.GLContextImpl.setGLFunctionAvailability(GLContextImpl.java:1942)
	at jogamp.opengl.x11.glx.X11GLXContext.createImpl(X11GLXContext.java:395)
	at jogamp.opengl.GLContextImpl.makeCurrentWithinLock(GLContextImpl.java:765)
	at jogamp.opengl.GLContextImpl.makeCurrent(GLContextImpl.java:648)
	at jogamp.opengl.GLContextImpl.makeCurrent(GLContextImpl.java:586)
	at com.jogamp.opengl.awt.GLJPanel$OffscreenBackend.initialize(GLJPanel.java:1731)
	... 59 more
java.lang.reflect.InvocationTargetException
	at java.desktop/java.awt.EventQueue.invokeAndWait(EventQueue.java:1367)
	at java.desktop/java.awt.EventQueue.invokeAndWait(EventQueue.java:1342)
	at java.desktop/javax.swing.SwingUtilities.invokeAndWait(SwingUtilities.java:1480)
	at com.jogamp.opengl.util.AWTAnimatorImpl.display(AWTAnimatorImpl.java:98)
	at com.jogamp.opengl.util.AnimatorBase.display(AnimatorBase.java:452)
	at com.jogamp.opengl.util.FPSAnimator$MainTask.run(FPSAnimator.java:178)
	at java.base/java.util.TimerThread.mainLoop(Timer.java:556)
	at java.base/java.util.TimerThread.run(Timer.java:506)

### Steps to Reproduce the Problem

  1. Open the program
  1. Click on Visualizer tab

### Specifications
#### Version

Product Version: Universal Gcode Platform 20200419
Java: 14; OpenJDK 64-Bit Server VM 14+36
Runtime: OpenJDK Runtime Environment 14+36
System: Linux version 5.6.5-arch3-1 running on amd64; UTF-8; en_US (ugsplatform)
User directory: /home/remi/.ugsplatform/2.0-SNAPSHOT/dev
Cache directory: /home/remi/.ugsplatform/2.0-SNAPSHOT/dev/var/cache

# glxinfo | grep version
server glx version string: 1.4
client glx version string: 1.4
GLX version: 1.4
    Max core profile version: 4.6
    Max compat profile version: 4.6
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.2
OpenGL core profile version string: 4.6 (Core Profile) Mesa 20.0.4
OpenGL core profile shading language version string: 4.60
OpenGL version string: 4.6 (Compatibility Profile) Mesa 20.0.4
OpenGL shading language version string: 4.60
OpenGL ES profile version string: OpenGL ES 3.2 Mesa 20.0.4
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
    GL_EXT_shader_implicit_conversions, GL_EXT_shader_integer_mix,

# archlinux-java status
Available Java environments:
  java-10-openjdk
  java-11-adoptopenjdk
  java-11-openjdk
  java-13-openjdk
  java-14-adoptopenjdk (default)
  java-8-openjdk/jre
#### Operating system

Linux - Archlinux

Yinameah, 2020-04-25:
I'm experiencing the same issue with UGS Classic.

I'm also running an up-to-date archlinux, same infos than above, except my default java is java-8-openjdk/jre, but I try with others environment and got similar issues.

I can send more detailled information about my system if usefull

lllars, 2020-06-16:
Same issue here.  Also on Arch Linux.  I'm using UGS Platform.  Anyone find a fix yet?

breiler, 2020-06-23:
I can reproduce the error with the JDK shipped by ArchLinux. It works when I use the linux version with a bundled JDK:
https://github.com/winder/Universal-G-Code-Sender#downloads

lllars, 2020-06-24:
Thanks for looking into this.  I wish I could duplicate your success, but I still can't get it working on my end.  I just re-downloaded the ugs-platform nightly (from your link) and tried again.  I even deleted my .ugs and .ugsplatform directories, but I'm still getting the same GL4bc error.  I can upload my messages.log if that would be useful.

I have also tried compiling ugs with maven, and am seeing the same error there.

lllars, 2020-06-24:
I just got this working by running ugs on my discrete video card instead of the intel integrated graphics.

`primusrun bin/ugsplatform`

It looks like this could be an Optimus problem.  I found another project that might have found a fix: https://github.com/runelite/runelite/pull/6530

closed 2020-07-17 by breiler

rainwoodman, 2021-12-28:
```LIBGL_ALWAYS_SOFTWARE=1 ./ugsplatform```
worked for me. I got this hint from: 

https://www.reddit.com/r/linuxquestions/comments/ll7fey/trying_to_setup_gephi_on_fedora_33_and_having/

This helps on Fedora 33's jdk-1.8.0 and on the downloaded JDK in the platform package (13). But I get a crash with Fedora 33's jdk-11.