winder/Universal-G-Code-Sender #1396: joystick doesn't work if reconnected

jon-daemon, 2020-04-23:
## Problem description 
I tried the nightly build and it works fine with a gamepad joystick. The porblem occurs if I disconnect the joystick or start UGS before connecting it. The joystick is working only if its connected before starting UGS. Is this normal?

### Steps to Reproduce the Problem
1. Connect the joystick
2. Open UGS (joystick is working fine)
3. remove joystick and reconnect. It stops working
or
1. Open UGS
2. Connect the joystick. Joystick is not working

### Specifications
#### Version
UGS Platform 2.0 - Nightly Build Apr 19, 2020

#### Operating system
Linux Mint

#### Platform
GRBL 1.1f

abdset, 2020-05-01:
Sounds like a bug that needs to be fixed.

mmosimann, 2022-09-13:
I'm facing the same problem using a xBox wireless joystick.

breiler, 2022-09-13:
@mmosimann does this problem still exist in version 2.0.12?

breiler, 2023-01-16:
Closing as this was fixed in #1927

closed 2023-01-16 by breiler