winder/Universal-G-Code-Sender #1397: UGS program stop working after M5 command

Envoy22, 2020-04-23:
Hi, i'm looking for a explanation/solution to my problem. The UGS software stop working after the M5 command. I tested for every UGS version disponible to download on website and the problem persists. Codes and commands run sucessifully but, in the end of sending, comes the M5 command to shut down the spindle. UGS returns the <ok> response, the spindle shuts down and program stop working. Extra: inicialy i used a i5 notebook, 16GB RAM, and this problem never occourred . Now i built a desktop with pentium processor and 4GB RAM. Could be there any problem with the PC?

abdset, 2020-05-01:
What do you mean by stop working, is it freezing or shuting down ?
Try to upgrade to grbl 1.1h.
Could you use the notebook again just to be sure the board is fine ?

Envoy22, 2020-05-01:
> What do you mean by stop working, is it freezing or shuting down ?
It stop working literaly. It shows a "disconnected" mensage and if i try to click anything, does not respond. After some seconds windows send the mensage "the program stop working". 

> Try to upgrade to grbl 1.1h.
grbl its upgraded to 1.1

> Could you use the notebook again just to be sure the board is fine ?
I used the notebook several times and never it presented this problem. And this days I have to use the notebook to avoid the problem envolving the PC. 

I guess it have something to do with eletric interference. I just observated that the problem occours when spindle turns on or off. So the notebook was working on batery, but the PC needs to be connected on the same energy source as the spindle.

link of the footage 
https://drive.google.com/file/d/11ix-QlMMaoby1pEqQCS8AqQ75DRb7ONR/view?usp=drivesdk

abdset, 2020-05-02:
- Do you have limit switchs installed ?
- Use cables with ground shield for the axis motors and connect the shield only to the GND of the controller, do not connect it to anything in the other side, this should resolve your problem.
- If you are using limit switchs you do the same for their wires but you will also need to add a capacitor 100 nF for each limit switch connected on its positive pin and the GND. You can also rise the value of $30 to prevent false triggering for the limit switchs.
- Keep in mind connecting any wire to ground should be done only on controller side, if you connect the other side to the machine body ground it will make it worst.

Envoy22, 2020-05-02:
> * Do you have limit switchs installed ?
> * Use cables with ground shield for the axis motors and connect the shield only to the GND of the controller, do not connect it to anything in the other side, [...]

Tanks! So grounding the cables only in controller side will reduce any eletric noises? I haven't any cables with groud shield. So I believe I should buy and remake all eletric conections. I will change the spindle cables too.
My limit switchs was accusing many false positives, so I disabled it. I will make a [RC filter](https://github.com/gnea/grbl/wiki/Wiring-Limit-Switches) in a PCB to resolve this.

closed 2020-07-17 by breiler