winder/Universal-G-Code-Sender #1399: UGS 4th axis

badassbiologist, 2020-04-29:
1. Can you control a 4th axis with UGS? 
2. If so what gcodes would be used to control simple movement?
3. Is there any interest in adding a 4th axis to the keyboard movement control on UGS?

Thanks

## Feature request

## Problem description 

### Expected Behavior

### Actual Behavior

### Steps to Reproduce the Problem

  1.
  1.
  1.

### Specifications
#### Version

#### Operating system

#### Platform

#### Other

abdset, 2020-05-01:
GRBL does not control 4 axis, same for UGS.
You can control 4 motors but one of them sould be cloned whitch makes it stay 3 axis.

breiler, 2022-07-12:
This should now be possible with GRBL-based controllers reporting extra axes in the status report, ex: `<Idle|MPos:1.000,2.000,3.000,4.000,5.000,6.0000|FS:0,0|Pn:XYZ>`

Controllers that I have tried:
* grblHAL
* FluidNC
* Grbl Mega 5x

If you find anything weird, please let us know.

closed 2022-07-12 by breiler