winder/Universal-G-Code-Sender #1400: UGS supporting UDP communication

villamvadasz, 2020-05-09:
## Feature request
UGS could support UDP/IP communication.

#### Platform
GRBL PIC32 port

breiler, 2020-11-16:
I think that using UDP is a bad idea. The protocol does not check if packets have been lost or come in the right order. You should probably want TCP communication.

closed 2020-11-16 by breiler

villamvadasz, 2020-11-21:
> Links:

Hi,

indead, I have observed some loss of packet. But for development it was 
really good, since I did not had to reconnect UGS all the time. I guess 
the a dedicated ethernet line would reduced the loss of packet to a 
level where it would go unnoticed.

Now I have changed my PIC32 GRBL to TCP and faced the following problem: 
UGS was not receiving message properly. It was stopping rigt after the 
connection, the CMD_RESET was sent out, and was received by my device. 
It answered back, but the answer was not processed. After a day, I have 
localised the problem, it was not my ethernet implementation, it was the 
UGS that was not receiving good.

My device answers with:
   printPgmString(("\r\nGrbl " GRBL_VERSION " ['$' for help]\r\n"));
There is a leading \r\n! I think this cause the problem. To be honest I 
never used BufferedReader or StreamWriter, so I do not have much 
experience with it. So I googled and found that other people use while 
loop. I guess to read out all the available lines. Not quite sure why a 
while loop is needed, since the thing runs in while loop already. But 
maybe the .ready() throws all previous processed lines away? So after a 
.ready the .readLine could return multiple lines? And when not read out 
all, with the next .ready call they are lost? Very strange. But who 
knows.

I think a while loop is missing from TCPConnection.java s function run:
				if (inStream.ready()) {
					while (1) {
						resp = bufIn.readLine());
						if (resp != null) {
							responseMessageHandler.handleResponse(resp + "\n");
						} else {
							break;
						}
					}
				}

After this UGS was working properly.

Did somebody tested the TCP part of UGS also with a string containing a 
leading \r\n? I guess without a leading \r\n the UGS would work. But at 
this point the leading \r\n is needed, since it is not ensured that the 
reset does not happen in a middle of a string sending. So the previous 
string might not ended with a \r\n.

The whole run function:
	public void run() {
		String resp;
		while(!Thread.interrupted() && !client.isClosed())
		{
			try {
				if (inStream.ready()) {
					while (1) {
						resp = bufIn.readLine());
						if (resp != null) {
							responseMessageHandler.handleResponse(resp + "\n");
						} else {
							break;
						}
					}
				}
			} catch (SocketException e) {
				e.printStackTrace();
				return; // terminate thread if disconnected
						//TODO: at some point, reconnecting should be considered
			} catch (IOException e) {
				e.printStackTrace();
				return; // terminate thread if disconnected
						//TODO: at some point, reconnecting should be considered
			}
		}
	}

Thank You!

2020-11-16 21:51 időpontban Joacim Breiler ezt írta:

villamvadasz, 2020-11-26:
Hi,

I think I made some mistake. I was trying different solutions and it 
could be that the one I sent was never compiling.

The first solution was this:
				//int ch = bufIn.read();
				//if (ch != -1) {
				//	receiveBuffer[receiveBufferCnt] = (byte)ch;
				//	receiveBufferCnt++;
				//	if (receiveBufferCnt >= 1492) {
				//		receiveBufferCnt = 0;
				//	} else {
				//		if ((ch == '\r') || (ch == '\n')) {
				//			resp = new String(receiveBuffer, 0, receiveBufferCnt);
				//			responseMessageHandler.handleResponse(resp + "\n");
				//			receiveBufferCnt = 0;
				//		}
				//	}
				//}
This was working and then I changed to the one that I sent. But today, I 
noticed that it does not even compile.
Now I fixed it and it compiles and works:
				if (inStream.ready()) {
					while (true) {
						resp = bufIn.readLine();
						if (resp != null) {
							responseMessageHandler.handleResponse(resp + "\n");
						} else {
							break;
						}
					}
				}
Not sure what exactly happend. I am not familiar with the Maven build 
environment, maybe there I made something wrong and I was wrongly 
thinking it compiled my changes.

Thank You!

2020-11-21 12:10 időpontban Andras Huszti ezt írta: