winder/Universal-G-Code-Sender #1401: UGS Shuts down when trying to connect to Grbl running on Arduino

wpong18082, 2020-05-18:
Hi there,

I am a newbie to UGS and am having an issue which I hope you could be of assistance.
I am trying to resurrect an old sony laptop running windows XP SP3.
The following has been installed 
UGS Classic Version: UniversalGcodeSender-v2.0.0-Aug.14.2019
Java version: jre1.8.0_74_x86
Grbl V1.1 running on Arduino Uno
Arduino IDE 1.8.9

I have made sure that I can see the Arduino Uno running Grbl is fine using the serial monitor in Arduino IDE and see the Grbl values by typing $$.
I can start UGS, it loads the UGS GUI okay, however when I select the correct comm port and the baud rate 115200 and try to connect to the Arduino Uno by selecting open, UGS classic just shutdown.

The log file was generated and is attached. 
[hs_err_pid1304.log](https://github.com/winder/Universal-G-Code-Sender/files/4647272/hs_err_pid1304.log)

I have tried restarting the laptop and updating the Java version runtime and the same issue occurs.

I have a similar set up on a window 7 laptop running a slighly version of Java and this seem to work fine. 

Best regards,
Wilder

abdset, 2020-06-29:
I'm not sure where to activate minidump in windows XP  (Yours is disabled somehow) but in newer versions of windows you can activate that by allowing some amount of disk space to be a virtual memory so the system can use it when it runs out of physical memory.
Also you may run the program as administration (dont remember is this is doable in win XP).

Since May 2019 release, windows 10 is able to be installed in a very old and poor PC configs now, you could try that and definitly fix your problem, i'm using a core 2 duo machine, 4Gb of ram, running win 10 and working like a charm.

Here more explaination about minidump file:
https://en.wikipedia.org/wiki/Core_dump

Sorry for being late. Good luck.

breiler, 2023-01-19:
We have upgraded JSerialComm, if this is still a problem please create a new issue.

closed 2023-01-19 by breiler