winder/Universal-G-Code-Sender #1403: estimated time

cornnatron, 2020-05-31:
## Feature request

breiler, 2020-06-01:
The estimation is done based on how many commands that has been processed in a given time and then use that to figure out how much time it should take to process remaining commands.
It is the a poor mans way to do it. I.e. it does not take into account if the the commands uses rapid movement or not. =)

The Gcode parsing has been improved a lot over the years so it could be possible to  make more accurate predictions analysing feed and length of movement.

cornnatron, 2020-06-01:
I understand how its doing its job figured that out after some searching yesterday thats why i asked if there was an option for it if not would be nice if an option could be added.
Someting like after loading the file it calculates the time but gives u the option to change it if wanted.

closed 2026-08-30 by breiler